realsenseai/librealsense #2248: .NET Cannot use camera after unplug/plug

mknejp, 2018-08-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.15.0.163                          | 
| Language                            |  C# | 

Given the following piece of code
```csharp
void Stream()
{
  using (var ctx = new Context())
  using (var pipe = new Pipeline(ctx))
  {
    var config = new Config();
    config.DisableAllStreams();
    config.EnableStream(Stream.Color, 1280, 720, Format.Bgr8, 30);
    config.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
    pipe.Start(config);

    while (run)
    {
      using (var frame = pipe.WaitForFrames())
      {
        Handle(frame);
      }
    }
  }
}
```

When the camera is unplugged while this loop is running we get an error:
```
Exception thrown: 'System.Exception' in Intel.RealSense.dll
Device disconnected. Failed to recconect: No device connected5000
rs2_pipeline_wait_for_frames(pipe:06413080)
```
This is expected. Once the camera is plugged in again we call the above code again but it fails at `pipe.Start()` with
```
Exception thrown: 'System.Exception' in Intel.RealSense.dll
_source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: "No such interface supported"
rs2_pipeline_start_with_config(pipe:07D69BD0, config:07D69B70)
```
This is a regression from 2.13 where calling the above after plugging the camera back in would restart the video stream. Is this intentional? Is there some other way now to continue a stream after reconnecting the camera?

RealSense-Customer-Engineering, 2018-08-21:
[Realsense Customer Engineering Team Comment]
Hi @mknejp,

Try to use "try...catch" to catch the exception, then have pipe.Stop() and pipe.Release().
You should be able to re-run the pipe config and streams.

mknejp, 2018-08-22:
I don't have access to the camera anymore for a couple weeks so I can't test this right now, but looking at the implementation I see that `Pipeline.Dispose()` calls `Pipeline.Release()` and as `pipe` is declared inside a `using` block that should get called when the exception gets thrown.

And as mentioned this used to work in 2.13.

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
@ev-mp, @dorodnic,
any more comments about the implementation between 2.15 and 2.13?

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
Hi @mknejp,

Do you have a chance to try?

RealSense-Customer-Engineering, 2018-11-09:
[Realsense Customer Engineering Team Comment]
Hi @mknejp,

Would you also like to try the latest librealsense to see how it works?

Gallimathias, 2018-12-04:
Hi guys, I have tested this bug with the latest firmware and Realsense version. The bug doesn't occur anymore.

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Hi @mknejp,

Do you have the same positive result as @Gallimathias verified?

mknejp, 2018-12-12:
Yes, @Gallimathias is actually the person who has taken over the project since I no longer have access to the camera. Sorry for being so sparse on communication.

closed 2018-12-12 by RealSense-Customer-Engineering