realsenseai/librealsense #2249: A few questions about sdk & Viewer:

joelihn, 2018-08-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| Language                            |  {C++ }                          | 
| Viewer Version                   |  2.13              | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, i have 3 questions about the sdk & viewer:

1).When i use rs2::software_device instance, may use register_extrinsics_to function register A device`s depth stream to B device`s color stream？If it is ok,when i add the software device source in Viewer, does it mean that the textrue source and depth source in 3D view will show correctly?

2)Is Viewer support record the Platform Camera as normal video format files, like mp4 etc..

3).May i open a avi or mp4 file , and put it into software_device`s color stream?

Thanks !

dorodnic, 2018-08-16:
Hi @joelihn 

> 1).When i use rs2::software_device instance, may use register_extrinsics_to function register A devices depth stream to B devices color stream？

Yes, it is possible to add extrinsics / intrinsics for every two stream and all SDK processing should work with it. 

> If it is ok,when i add the software device source in Viewer, does it mean that the textrue source and depth source in 3D view will show correctly?

Currently, there is no built-in way to add custom software device to the Viewer. This can be done, by adding `add_software_device` function to the `context`.

> 2)Is Viewer support record the Platform Camera as normal video format files, like mp4 etc..

We are using uncompressed ROS-bag format for all recording. Standard formats rely on codec libraries like FFMPEG, but these come with licensing limitations that we prefer to avoid.  

> 3).May i open a avi or mp4 file , and put it into software_device`s color stream?

You can read frames using something like OpenCV and send them to software device.

joelihn, 2018-08-19:
Below is my code, i have a xml config record devices stream extrinsics like this:

`<Config>
    <Device SN="A">
    		 <Color SN="A">
        	<MT>1.0000 0.0038 -0.0009 -0.0038 1.0000 0.0058 0.0008 -0.0058 1.0000 -0.025.6999989 0.000733261108 -0.00396292090</MT>
    		</Color>
        <Color SN="B">
        	<MT>1 0 0 0 1 0 0 0 1 0 0 0</MT>>   
    		</Color>
    </Device>
    <Device  SN="B">
    	 <Color SN="A">
        	<MT>1 0 0 0 1 0 0 0 1 0 0 0</MT>>
    		</Color>
       	<Color SN="A">
    			<MT>1 -0.0038 0.0008 0.0038 1 -0.0058 -0.0009 0.0058 1 -0.0300679 -0.000661246 0.00393617</MT>>
    		</Color>
    </Device>
</Config>`

in ds5-color.cpp`s constructor function i regisiter the extrinsics for every device depth stream to every device color stream .

when i use the use the depth1&color1 or depth2&color2 , i change the extrinsics has affect, but when i use depth1&color2 or depth2&color1, thers is no affect, so i dont know sth i doing wrong,thanks.

`// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <librealsense2/hpp/rs_internal.hpp>
#include "example.hpp"
#include "example.hpp"          // Include short list of convenience functions for rendering
#include <windows.h>

#define STB_IMAGE_WRITE_IMPLEMENTATION
#include <stb_image_write.h>
#include <int-rs-splash.hpp>

#ifndef STB_IMAGE_IMPLEMENTATION
#define STB_IMAGE_IMPLEMENTATION 
#include <stb_image.h>
#endif // !1

#include <iostream>

int main(int argc, char * argv[]) try
{
	window app(1920, 1080, "RealSense Capture Example");
	glfw_state app_state;
	register_glfw_callbacks(app, app_state);

	rs2::pointcloud pc;
	rs2::points points;
	int frame_number = 0;

	//////////////////////////////////////////////////////////////////////////////////////
	// Using the context we can get all connected devices in a device list
	rs2::context ctx;
	rs2::device_list devices = ctx.query_devices();
	std::cout << "devices cout is: " << devices.size() << std::endl;
	int index = 0;

	rs2::pipeline p0;
	rs2::config c0;
	std::cout << "devices[0] SERIAL_NUMBER is: " << devices[0].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
	c0.enable_device(devices[0].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
	// Start the pipeline with the configuration
	rs2::pipeline_profile profile0 = p0.start(c0);

	rs2::pipeline p1;
	rs2::config c1;
	std::cout << "devices[1] SERIAL_NUMBER is: " << devices[1].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
	c1.enable_device(devices[1].get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
	// Start the pipeline with the configuration
	rs2::pipeline_profile profile1 = p1.start(c1);

	auto depth_stream1 = profile0.get_stream(RS2_STREAM_DEPTH);
	auto color_stream1 = profile0.get_stream(RS2_STREAM_COLOR);

	auto depth_stream2 = profile1.get_stream(RS2_STREAM_DEPTH);
	auto color_stream2 = profile1.get_stream(RS2_STREAM_COLOR);

	//==================================================//
	//           Declare Software-Only Device           //
	//==================================================//

	rs2::software_device dev; // Create software-only device
	auto depth_sensor = dev.add_sensor("Depth"); // Define single sensor
	auto color_sensor = dev.add_sensor("Color"); // Define single sensor

	dev.create_matcher(RS2_MATCHER_DLR_C);
	rs2::syncer sync;

	depth_sensor.open(depth_stream2);
	color_sensor.open(color_stream1);

	depth_sensor.start(sync);
	color_sensor.start(sync);

	//depth_stream2.register_extrinsics_to(color_stream1, { { 1,0,0,0,1,0,0,0,1 },{ 0,0,0 } });

	while (app) // Application still alive?
	{
		rs2::frameset frameset1 = p0.wait_for_frames();
		rs2::frameset frameset2 = p1.wait_for_frames();

		++frame_number;

		rs2::frame depth1 = frameset1.first_or_default(RS2_STREAM_DEPTH);
		rs2::frame color1 = frameset1.first_or_default(RS2_STREAM_COLOR);
		rs2::frame depth2 = frameset2.first_or_default(RS2_STREAM_DEPTH);
		rs2::frame color2 = frameset2.first_or_default(RS2_STREAM_COLOR);

		if (**depth1** && **color2**)
		{
			if (auto as_depth = **depth1**.as<rs2::depth_frame>())
				points = pc.calculate(as_depth);
			pc.map_to(**color2**);

			// Upload the color frame to OpenGL
			app_state.tex.upload(**color2**);
		}
		draw_pointcloud(app.width(), app.height(), app_state, points);
		Sleep(1000);
	}

	return EXIT_SUCCESS;
 }catch (const rs2::error & e)
{
	std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
	return EXIT_FAILURE;
}
catch (const std::exception& e)
{
	std::cerr << e.what() << std::endl;
	return EXIT_FAILURE;
}

`

joelihn, 2018-08-24:
Someone can help me check this question？Thanks in advance.

ev-mp, 2018-08-28:
Hello @joelihn,
That's an interesting use-case indeed, and cross-device registration is possible.
There was a single line in the code that had to be fixed to make it work :
> depth_stream**2**.register_extrinsics_to(color_stream**1**, { { 1,0,0,0,1,0,0,0,1 },{ 0,0,0 } });

to 

> depth_stream**1**.register_extrinsics_to(color_stream**2**, { { 1,0,0,0,1,0,0,0,1 },{ 0,0,0 } });

:)

 Additionally, the point cloud calculation API should be reordered:

>**pc.map_to(color2);**    // This is the appropriate place for the call
if (auto as_depth = depth1.as<rs2::depth_frame>())
             points = pc.calculate(as_depth);
~~pc.map_to(color2);~~ // At this point has no effect as the pointcloud and [u,v] map are already generated

This error would manifest itself clearly if the pointcloud was generated once and not in a loop

joelihn, 2018-09-14:
@ev-mp ok,thank you very much!

I have another question , may i add a software device into devicelist in Intel RealSense Viewer? Is there some sample code?

dorodnic, 2018-09-21:
hi @joelihn 
Sorry for late response, we are a bit swamped temporarily. Right now there is no `add_device` API for the `context` class. I did some work in this direction some time ago. You can take a look at [dorodnic/adv_software_device](https://github.com/dorodnic/librealsense/tree/adv_software_device) branch, specifically [this commit](https://github.com/dorodnic/librealsense/commit/b08d62c0e38f7d684167454542a0d80ada048961). The branch requires some polish, but it shows how you can run python opencv inside the Viewer, and was used to generate the visualisations for [stereo depth blog post](https://realsense.intel.com/stereo-depth-vision-basics/). At the time, none showed particular interest in software device APIs, but perhaps its time to refine and add to the main SDK.

joelihn, 2018-10-08:
Thanks for ur response. I think it is very useful add software device into viewer.
wait in hope.

closed 2019-02-27 by RealSenseCustomerSupport