realsenseai/librealsense #2255: Hi,

ankith001, 2018-08-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-08-17:
[Realsense Customer Engineering Team Comment]
hi @ankith001,

yes, RGB and depth are separate camera so different frame rate for depth and RGB is possible.

RealSense-Customer-Engineering, 2018-08-30:
[Realsense Customer Engineering Team Comment]
hi @ankith001,

do you need more support on this topic?

upxela, 2018-09-24:
I'm experiencing a similar issue on D435: when I stream RGB, ir, and Depth at 30 fps at 1280x720, I realize that I do not achieve the desired framerate. Streaming 5 minutes of video at 30fps should yield approximately 9000 frames, but I only achieve 6204 depth frames, 5834 ir frames, and 6180 rgb frames. When I run either only depth and ir or only rgb, I get near the full framerate. 

My code is as follow:
`length = 1280`
`width = 720`
`fps = 30`
`pipeline = rs.pipeline()`
`config = rs.config()`
`config.enable_stream(rs.stream.depth, length, width, rs.format.z16, fps)`
`config.enable_stream(rs.stream.infrared, length, width, rs.format.y8, fps)`
`config.enable_stream(rs.stream.color, length, width, rs.format.rgb8, fps)`
`config.enable_record_to_file('./path2bag')`
`try:`
	`pipeline.start(config)`
	`time.sleep(60*5)`
`finally:`
	`pipeline.stop()`

RealSense-Customer-Engineering, 2018-09-25:
[Realsense Customer Engineering Team Comment]
Hi @upxela,

The similar issue with #2216. Checking the fix.

ankith001, 2018-09-25:
I still don’t know why do we get two different answers from realsense team.
We still are facing the issue of different frames. Please verify and let us
know about the issue.

RealSense-Customer-Engineering, 2018-09-25:
[Realsense Customer Engineering Team Comment]
hi @ankith001,

The previous response is for the bag record/playback issue.
To further clarify your issue about the difference between depth FPS and RGB FPS, can you let me know how large FPS difference you see? which librealsense version and firmware do you use?

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @ankith001,

Another check is to verify the impact of the exposure. See #2436.

ankith001, 2018-09-26:
Hi,

I see around 7-10 FPS difference between depth and RGB FPS.

RealSense-Customer-Engineering, 2018-09-27:
[Realsense Customer Engineering Team Comment]
Hi @ankith001,

You can control RGB AE or Auto-Exposure Priority to see how it works.

ankith001, 2018-09-28:
I have enabled Auto Exposure-Priority. But still, I face the decrease in my
FPS both in depth and RGB. It's 1 in 50 times I get 30 FPS for depth and
RGB.

Thanks,
Ankith

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
Hi @ankith001,
To keep the required FPS, you can try below:
1. This can be forced by disabling Exposure Priority control (this way auto-exposure will not reduce RGB FPS).
2. Disable auto exposure.

ankith001, 2018-10-02:
Thanks, How to ensure I get constant FPS even in Depth. Because the FPS in
depth also varies

Thanks,
Ankith

RealSense-Customer-Engineering, 2018-10-03:
[Realsense Customer Engineering Team Comment]
Hi @ankith001,

What's your test scene to have the varying FPS for Depth and Color?
You can also disable auto exposure for depth to keep FPS. But, you need to manually control the exposure value according to your test scene.

RealSense-Customer-Engineering, 2018-10-15:
[Realsense Customer Engineering Team Comment]
Hi @ankith001,

How is your test result? Is the issue clarified?

closed 2018-11-06 by RealSense-Customer-Engineering