realsenseai/librealsense #2257: Align Depth to Color Image Quite Slow on ARM

connorsoohoo, 2018-08-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.38)                                         | 
| Platform                                 | NVIDIA Jetson Tx2 |
| SDK Version                            |  { 2.14.1 }                          | 
| Language                            |  { C++ }                          | 
| Segment			|  { Robot }                          | 

### Issue Description

I want to be able to output an aligned color and depth frame. However, for ARM architectures, it takes more than 50 milliseconds to perform an alignment. It looks like the optimizations that you made to speed up this alignment are only available if the hardware supports the SSSE3 instruction set. Because the Jetson Tx2 is an ARM device, it does not benefit from this optimized version. Is there another faster way of aligning the depth to the color frame perhaps using CUDA?

RealSense-Customer-Engineering, 2018-08-17:
[Realsense Customer Engineering Team Comment]
Hi @connorsoohoo,

librealsense 2.13.0 (or later release) includes CUDA-optimized implementation for Jetson-TX (arm) platform. You can have a try to enable the build option "BUILD_WITH_CUDA".

dorodnic, 2018-08-19:
`align` post-processing block still does not take advantage of CUDA, right now only color conversions and point-cloud generation are CUDA-optimized. We would love to add CUDA align, but it can take some time until we get to it.