realsenseai/librealsense #2258: How to run the examples on RealSense D435?

arya001, 2018-08-17:
---------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.4.0-131                    | 
| Platform                                 | laptop |
| SDK Version                            |   v2.15.0    | 
| Language                            |  CPP    | 

### Issue Description

Hi, Sorry for asking this basic a question.

I installed the librealsense according to the instructions.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

However I cant seem to run the examples. Any help will be appreciated.

RealSense-Customer-Engineering, 2018-08-17:
[Realsense Customer Engineering Team Comment]
hi @arya001,

can  you provide more detail about what error or issue you see?
thanks.

arya001, 2018-08-17:
@RealSense-Customer-Engineering 

Hi, 
After I posted the query, I downloaded the librealsense-master folder from github. I ran `user:~/librealsense-master$ cmake .` and `user:~/librealsense-master$ make` .

After that I could run all the example files. So I think it is fine now. I guess I should close the issue.

closed 2018-08-17 by arya001

ZaidKamil, 2019-06-12:
Hello. I am still confused on how you were able to run 
user:~/librealsense-master$ cmake . and user:~/librealsense-master$ make on Github