realsenseai/librealsense #2259: Capturing left and right frames at 16bpp

lpuglia, 2018-08-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Linux (Fedora 26)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.13.9)                                         | 
| Platform                                 | PC |

### Issue Description
When i try to connect to the board with the following configuration:

[image]

the connection button get disabled, I am pretty sure i read somewhere that the left and right cameras are set to capture frame at `>8bit per pixel`, but apparently it is not possible to capture those raw frames from the viewer. Is it possible to get those raw images from the API?

Regards

MartyG-RealSense, 2018-08-17:
There was a discussion recently over at the Intel Support site about why Y16 mode is unavailable for most settings.  

https://communities.intel.com/message/557377#557377

The short version of the above discussion is that Intel advises against using Y16 in an application because it is unrectified and intended for use in camera calibration.  UYVY infrared mode may work better than Y16 for you.

timprepscius, 2018-08-21:
The Y16 is restricted with regard to FPS.  You can access it if you figure out the correct FPS.
Use the "enumerate-devices" tool to find out which parameters will allow you to access the Y16 streams.

lpuglia, 2018-08-21:
I see, anyway the stream is unrectified, not very useful for my purposes.

Out of curiosity how do you calibrate each pair of camera for each device? is it done automatically at boot?

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
hi @lpuglia,

we calibrate the camera peripheral in our factory, and end user can use dynamic calibration to calibrate in the field.
please refer to https://realsense.intel.com/intel-realsense-downloads/#cal

closed 2018-11-29 by dorodnic