realsenseai/librealsense #2260: Can I use c++'s multithreading to develop the realsense D400 family?

PipgZ, 2018-08-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 16)  |  
| SDK Version                            |  { legacy / 2.<14>.<1> }                          | 
| Language                            |  {c++}                          | 

### Issue Description
<When I use c++'s multithreading to develop the realsense D400 family, it show show the error about that I can not open the camera.>

dorodnic, 2018-08-19:
Hi @PipgZ 
The high-level answer is yes, there are several samples using threads.
Please share more information about what you are doing, so we can help.
What operating system are you using? What error do you get? Can you share the code snippet?

RealSense-Customer-Engineering, 2018-08-31:
[Realsense Customer Engineering Team Comment]
@PipgZ
Any update?

closed 2018-11-06 by dorodnic