realsenseai/librealsense #2261: rs-measure example does not align depth and color part

kubrakaracan, 2018-08-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (05.09.14.00) | 
| Operating System & Version |   { Linux (Ubuntu 16)}  | 
| Kernel Version (Linux Only)    |  (4.15.32)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.15 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi. rs-measure does not work properly, it does not align depth image with color part. I assume that it occurred after updating firmware from 2.13 to 2.14 (because it was working in previous versions). Now I updated my camera to 2.15 but the problem still exists. Thanks in advance  
[image]
  >

RealSense-Customer-Engineering, 2018-08-21:
[Realsense Customer Engineering Team Comment]
hi @kubrakaracan,

will have a try later but would like to check if you try dynamic calibration for this issue?
thanks

kubrakaracan, 2018-08-24:
Actually, I meant to say that in previous versions the example gave me the view below. Now, the view has not the color part. However, your view also has not the color part. What is the reason for that ? Is it a normal situation ? 
[image]

temmat, 2018-08-24:
I'm dealing with a similar problem and here's what I found out...

@kubrakaracan I apologize if what I'll describe is not what you see / your issue, but I think it is judging by your last comment. I don't want to take over your thread/issue, just want to add to it since I think I found the same thing as you did and I did some digging.

@RealSense-Customer-Engineering @JasonChang13 no (at least in my case, I can't be sure for kubrakaracan), but it has nothing to do with calibration or closeness of the object. The "color" frame when rendered (the code of line doing this is color_image.render(color, { 0, 0, app.width(), app.height() }); ) actually renders to....nothing, or to a black frame to be precise. You can verify this by either disabling glBlend or don't even render the depth frame and just output the color one at this point. (glBlendFunc setup doesn't change this.)

It gets interesting (in the sense that you do get a rendered color frame) though if you:
-disable frame alignment by disabling "data = align_to.process(data);" in the video_processing_thread (but in this case of course the color frame is not aligned to anything)
OR
-change the color data stream from RS2_FORMAT_RGBA8 to RS2_FORMAT_RGB8 (but then you have to fiddle either with glBlendFunc or glColor4f to see both rendered frames).

I'm not sure because I haven't been able to check yet how well the aligning process works (I've read the docs, but haven't gone through the code yet), but it looks like there's something going on with the aligning process when it's dealing with a stream that has an alpha channel (or an alpha channel with the contents that the Realsense RGBA8 stream provides).

-------------------------------------------------------------------

Added later:
Yep, that's exactly what https://github.com/IntelRealSense/librealsense/pull/2264 deals with; the fourth byte in RGBA8.

Check out the change / commit to see if it helps you. It's not included in the 2.15 release (latest windows stable release).

dorodnic, 2018-09-06:
I think we should be good now. 
Thank you for feedback.

closed 2018-09-06 by dorodnic