realsenseai/librealsense #2262: SR300 Detected unvalidated firmware and could not get RVis image

mambawockeez, 2018-08-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300 } | 
| Firmware Version                   | (3.10.10.0) | 
| Operating System & Version |   {Linux (Ubuntu 16) }  | 
| Kernel Version (Linux Only)    |  (latest)                                         | 
| Platform                                 | PC  |

### Issue Description
<Hi im trying to learn ROS and im using razer stargazer(realsense sr300). But when I run roslaunch realsense_camera sr300_nodelet_default.launch it tells me [ WARN] [1534628187.488155464]: /camera/driver - Detected unvalidated firmware:- 61020100024 's current camera firmware is 3.21.0.0, Validated camera firmware is 3.10.10.0 and when im using RVis, it doesnt allow me to choose size and color. 
[image]
Then im tring to reinstall realsense package, but it always trying to find file to fetch.
[image]

/ so please tell me how to solve my problem, i really appreciated your help! >

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
hi @mambawockeez,

would like to know which librealsense version you are using?

mambawockeez, 2018-08-23:
@RealSense-Customer-Engineering and Im use razer's stargazer camera, and i could get RViz image by executing 'roslaunch realsense_camera r200_nodelet_rgbd.launch' but not 'roslaunch realsense_camera r200_nodelet_default.launch'

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @mambawockeez,

It looks you use an old ROS wrapper which is not maintained. Please follow the instructions in below link to re-build the ROS wrapper which supports librealsense 2.x and it should work with SR300 + FW 3.21.0.
https://github.com/intel-ros/realsense

mambawockeez, 2018-09-07:
Thanks a lot I’ll try and see what will happen!

mambawockeez, 2018-09-18:
@RealSense-Customer-Engineering 
[image]
[image]

HI i dont know why when i follow the steps u gave me, i get this two error. im really new to this, could u plz tell me what is going on right here?

mambawockeez, 2018-09-18:
@RealSense-Customer-Engineering 
[image]
there is an error when i type in the first command
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade

RealSense-Customer-Engineering, 2018-09-19:
[Realsense Customer Engineering Team Comment]
Hi @mambawockeez,

You can refer to below to see if it can fix your issue.
https://askubuntu.com/questions/742221/chrome-updates-fail-with-error-when-updating-repository-information?rq=1

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
Hi @mambawockeez,

Is the issue resolved?