realsenseai/librealsense #2269: automatic exposure does not work for first few frames

waspinator, 2018-08-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.09.14.00| 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.13.0-41-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2                        | 
| Language                            |  python                       | 
| Segment			|  Robot                         | 

### Issue Description

When running the (opencv example)[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py] and replacing `imshow` with `imwrite` followed by a `break`, the first few frames are underexposed. 

A work around is to ignore the first 5 or so frames before saving an image. Is there something like `pipeline.wait_for_good_frames()` to prevent under/over exposed, or any other unusable frames from coming through the pipeline?

MartyG-RealSense, 2018-08-20:
Intel does a similar method to your workaround in a script in their Getting Started guide for OpenCV.  They drop the first few frames to allow the auto-exposure to settle, then use frames = pipe.wait_for_frames()

 // Camera warmup - dropping several first frames to let auto-exposure stabilize
    rs2::frameset frames;
    for(int i = 0; i < 30; i++)
    {
        //Wait for all configured streams to produce a frame
        frames = pipe.wait_for_frames();
    }

dorodnic, 2018-08-20:
Hi @waspinator 
It is possible to set up manual exposure ahead of time to the desired level. However, if you are using auto-exposure, it will take several frames to converge. 
Skipping first several frames is a decent rule of thumb, but if you want to be more scientific about it, you can read the actual exposure of each frame using `get_frame_metadata` and wait util it stabilizes. Just make sure you have the correct kernel patch applied (otherwise metadata will not work).

RealSense-Customer-Engineering, 2018-08-29:
[Realsense Customer Engineering Team Comment]
Hi @waspinator,

Still need any support for this topic?

closed 2018-08-29 by waspinator