realsenseai/librealsense #2271: Background Replacement

doseguritan, 2018-08-20:
Good Day,
I use the example of inside the wrappers/nodejs which is node-align,js
My question is how can i change the color gray to a transparent one.
If not possible how can i change the background using wrapper node js

doseguritan, 2018-08-23:
i am using d435

doseguritan, 2018-09-25:
hi @RealSense-Customer-Engineering 

i am using realsense depth camera d435.

below script come from node_align.js:

for (let i = 0; i < otherBpp; i++) {
        otherData[offset+i] = 0x99; // this one make the background gray. How can i make this transparent?
}

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
How about converting to RGBA to have the alpha channel for transparent parameter?

RealSense-Customer-Engineering, 2018-10-03:
[Realsense Customer Engineering Team Comment]
Another simple way is to blend the image with adding two images or with weighting. Just try below:
otherData[offset+i] += 0x99;

RealSense-Customer-Engineering, 2018-10-18:
[Realsense Customer Engineering Team Comment]
Still need any support for this topic? If not, will close this.

closed 2018-10-29 by RealSense-Customer-Engineering