realsenseai/librealsense #2272: Possible bug while storing frames

lpuglia, 2018-08-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Linux (Fedora 26)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.13.9)                                         | 
| Platform                                 | PC |

### Issue Description
The following snippet works with no problem:

```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>             // Terminal IO

int main(int argc, char * argv[])
{
	std::vector<rs2::frame> frames;
	rs2::pipeline pipe;
	auto profile = pipe.start();

	for (auto&& sensor : profile.get_device().query_sensors())
	{
	    sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE, 0);
	}

	for (int i = 0; i < 100; i++)
	{
	    rs2::frameset data = pipe.wait_for_frames();
	    frames.push_back(data.get_depth_frame());
	}

	for (auto&& frame : frames)
	    std::cout << frame.get_timestamp() << std::endl;
}
```

making small changes to store the whole `frameset` instead of a specific `frame`:

```
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <iostream>             // Terminal IO

int main(int argc, char * argv[])
{
	std::vector<rs2::frameset> frames;
	rs2::pipeline pipe;
	auto profile = pipe.start();

	for (auto&& sensor : profile.get_device().query_sensors())
	{
	    sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE, 0);
	}

	for (int i = 0; i < 100; i++)
	{
	    rs2::frameset data = pipe.wait_for_frames();
	    frames.push_back(data.get_depth_frame());
	}

	for (auto&& frame : frames)
	    std::cout << frame << std::endl;
}
```

returns:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 5000
```

Is this a documented behaviour?

dorodnic, 2018-08-20:
Hi @lpuglia 
Setting `RS2_OPTION_FRAMES_QUEUE_SIZE` to zero still works, but we eventually abandoned this approach in favor of a more generic API - [frame::keep()](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_frame.h#L208)
This way, you call `keep` on the frames you intend to save for more then a single iteration. If you intend to save the entire frameset you can also call `keep` on the frameset (I don't remember if you need to also call keep on the individual frames or not in this case)

lpuglia, 2018-08-20:
The fact is that i don't want to lose any frames, but even with this small workload (namely the `push_back`) i think some of the frames are dropped. What is the safest way to register, let's say, for 10 seconds without losing any frame?

timprepscius, 2018-08-21:
I would like to tag along here.  

Thread A grabs a frameset from pipe.wait_for_frames.
Thread A then sits for 2 minutes, yes 2 minutes.  All the while reading from the frame, over and over and over.
Thread A then calls pipe.wait_for_frames again.

Does Thread A need to call "keep" on that frame?  The examples seemed to suggest that the reference counting could do that for me.

I am seeing a sporadic crash within the usb code.  This is running in debug mode, and there is a chance that frame processing could take longer than the FPS allows for.

dorodnic, 2018-08-21:
Hi @timprepscius 
Feel free to share your code and we will try to help.
In general, frames are reference counted and can be freely shared between threads (we provide the `frame_queue` primitive to help with that)
The issue with `keep` is not strictly who holds a frame for how long. It is rather **how many** frames can you borrow from the SDK without breaking normal operation. We do everything with fixed pools to avoid per-frame allocations. When you save too many frames in your application, SDK pool will get depleted and you will stop getting new frames. That's where `keep` comes in.

RealSense-Customer-Engineering, 2018-08-31:
[Realsense Customer Engineering Team Comment]
@timprepscius
haven't got your response for a while, so I close the issue first, if you still have any question, feel free to re-open or open new issue.

closed 2018-08-31 by RealSense-Customer-Engineering