realsenseai/librealsense #2273: No color or fisheye data when starting the ROS node

jochemverboom, 2018-08-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-32-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.15.0                          | 
| Language                            | C++/ROS                         | 
| Segment			|  Robot                        | 

### Issue Description
I am currently configuring the launch file for the ros node. I am able to get infra1, infra2, depth but am unable to get the color and fisheye data. The camera is connected to a USB3.0 port using a USB3.0 cable (30 cm). What is wrong here?

When I start the ROS node I have the following output:
```

[1534757520.848485592][/camera/realsense2_camera_manager][Loader::Impl::advertiseRosApi]: Initializing nodelet with 8 worker threads.
[1534757520.869583993][/camera/realsense2_camera_manager][RealSenseNodeFactory::RealSenseNodeFactory]: RealSense ROS v2.0.3
[1534757520.869617942][/camera/realsense2_camera_manager][RealSenseNodeFactory::RealSenseNodeFactory]: Running with LibRealSense v2.15.0
[1534757520.986994803][/camera/realsense2_camera_manager][BaseRealSenseNode::getParameters]: getParameters...
[1534757521.055857808][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: setupDevice...
[1534757521.055886817][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: JSON file is not provided
[1534757521.055907451][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: ROS Node Namespace: camera
[1534757521.055927014][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Device Name: Intel RealSense D430
[1534757521.055942651][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Device Serial No: 827312070061
[1534757521.055958216][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Device FW version: 05.09.14.00
[1534757521.055973744][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Device Product ID: 0x0AD4
[1534757521.055989028][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Enable PointCloud: On
[1534757521.056002997][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Align Depth: On
[1534757521.056017079][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Sync Mode: On
[1534757521.056176363][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Device Sensors: 
[1534757521.056290955][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: Stereo Module was found.
[1534757521.056421438][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[1534757521.056441221][/camera/realsense2_camera_manager][BaseRealSenseNode::setupDevice]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[1534757521.056477323][/camera/realsense2_camera_manager][BaseRealSenseNode::setupPublishers]: setupPublishers...
[1534757521.058801910][/camera/realsense2_camera_manager][FrequencyDiagnostics::FrequencyDiagnostics]: Expected frequency for depth = 30.00000
[1534757521.062074395][/camera/realsense2_camera_manager][BaseRealSenseNode::setupStreams]: setupStreams...
 20/08 11:32:01,062 WARNING [140297040021376] (sensor.cpp:338) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[1534757521.063968326][/camera/realsense2_camera_manager][BaseRealSenseNode::enable_devices]: depth stream is enabled - width: 1280, height: 720, fps: 30
[1534757521.070513551][/camera/realsense2_camera_manager][BaseRealSenseNode::publishStaticTransforms]: publishStaticTransforms...
[1534757521.070636106][/camera/realsense2_camera_manager][BaseRealSenseNode::publishTopics]: RealSense Node Is Up!
 20/08 11:32:01,073 WARNING [140296356611840] (backend-v4l2.cpp:872) Empty frame has arrived.
 20/08 11:32:01,073 WARNING [140296356611840] (backend-v4l2.cpp:872) Empty frame has arrived.
 20/08 11:32:01,073 WARNING [140296356611840] (backend-v4l2.cpp:872) Empty frame has arrived.
 20/08 11:32:01,073 WARNING [140296356611840] (backend-v4l2.cpp:872) Empty frame has arrived.
```

Also the info from 'rs-enumerate-devices' tells me the following (no color or fisheye):

```
Device info: 
    Name                          : 	Intel RealSense D430
    Serial Number                 : 	827312070061
    Firmware Version              : 	05.09.14.00
    Recommended Firmware Version  : 	05.09.14.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD4
    Usb Type Descriptor           : 	3.2

Stream Profiles supported by Stereo Module
 Supported modes:                                      stream     resolution fps   format   
    Infrared 1	  1280x800	@ 30Hz	Y8
    Infrared 2	  1280x800	@ 30Hz	Y8
    Infrared 2	  1280x800	@ 25Hz	Y16
    Infrared 1	  1280x800	@ 25Hz	Y16
    Infrared 1	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y16
    Infrared 2	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x800	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 30Hz	Y8
    Infrared 2	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 15Hz	Y8
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 2	  1280x720	@ 6Hz	Y8
    Infrared 2	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 90Hz	Y8
    Infrared 1	  848x480	@ 60Hz	Y8
    Infrared 2	  848x480	@ 60Hz	Y8
    Infrared 1	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 30Hz	Y8
    Infrared 2	  848x480	@ 15Hz	Y8
    Infrared 1	  848x480	@ 15Hz	Y8
    Infrared 2	  848x480	@ 6Hz	Y8
    Infrared 1	  848x480	@ 6Hz	Y8
    Infrared 2	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 90Hz	Y8
    Infrared 1	  640x480	@ 60Hz	Y8
    Infrared 2	  640x480	@ 60Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 2	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 15Hz	Y8
    Infrared 2	  640x480	@ 6Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 1	  640x400	@ 25Hz	Y16
    Infrared 2	  640x400	@ 25Hz	Y16
    Infrared 1	  640x400	@ 15Hz	Y16
    Infrared 2	  640x400	@ 15Hz	Y16
    Infrared 1	  640x360	@ 90Hz	Y8
    Infrared 2	  640x360	@ 90Hz	Y8
    Infrared 1	  640x360	@ 60Hz	Y8
    Infrared 2	  640x360	@ 60Hz	Y8
    Infrared 2	  640x360	@ 30Hz	Y8
    Infrared 1	  640x360	@ 30Hz	Y8
    Infrared 2	  640x360	@ 15Hz	Y8
    Infrared 1	  640x360	@ 15Hz	Y8
    Infrared 2	  640x360	@ 6Hz	Y8
    Infrared 1	  640x360	@ 6Hz	Y8
    Infrared 1	  480x270	@ 90Hz	Y8
    Infrared 2	  480x270	@ 90Hz	Y8
    Infrared 2	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 30Hz	Y8
    Infrared 2	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Infrared 2	  480x270	@ 6Hz	Y8
    Infrared 1	  424x240	@ 90Hz	Y8
    Infrared 2	  424x240	@ 90Hz	Y8
    Infrared 2	  424x240	@ 60Hz	Y8
    Infrared 1	  424x240	@ 60Hz	Y8
    Infrared 2	  424x240	@ 30Hz	Y8
    Infrared 1	  424x240	@ 30Hz	Y8
    Infrared 1	  424x240	@ 15Hz	Y8
    Infrared 2	  424x240	@ 15Hz	Y8
    Infrared 1	  424x240	@ 6Hz	Y8
    Infrared 2	  424x240	@ 6Hz	Y8
    Depth	  1280x720	@ 30Hz	Z16
    Depth	  1280x720	@ 15Hz	Z16
    Depth	  1280x720	@ 6Hz	Z16
    Depth	  848x480	@ 90Hz	Z16
    Depth	  848x480	@ 60Hz	Z16
    Depth	  848x480	@ 30Hz	Z16
    Depth	  848x480	@ 15Hz	Z16
    Depth	  848x480	@ 6Hz	Z16
    Depth	  640x480	@ 90Hz	Z16
    Depth	  640x480	@ 60Hz	Z16
    Depth	  640x480	@ 30Hz	Z16
    Depth	  640x480	@ 15Hz	Z16
    Depth	  640x480	@ 6Hz	Z16
    Depth	  640x360	@ 90Hz	Z16
    Depth	  640x360	@ 60Hz	Z16
    Depth	  640x360	@ 30Hz	Z16
    Depth	  640x360	@ 15Hz	Z16
    Depth	  640x360	@ 6Hz	Z16
    Depth	  480x270	@ 90Hz	Z16
    Depth	  480x270	@ 60Hz	Z16
    Depth	  480x270	@ 30Hz	Z16
    Depth	  480x270	@ 15Hz	Z16
    Depth	  480x270	@ 6Hz	Z16
    Depth	  424x240	@ 90Hz	Z16
    Depth	  424x240	@ 60Hz	Z16
    Depth	  424x240	@ 30Hz	Z16
    Depth	  424x240	@ 15Hz	Z16
    Depth	  424x240	@ 6Hz	Z16
```

dorodnic, 2018-08-20:
Hi @jochemverboom 
What type of camera do you have? Based on the logs, it is not supposed to have color or fish-eye streams.

jochemverboom, 2018-08-20:
The camera I have is an Intel D430 sensor module combined with the Intel D4 Board.

jochemverboom, 2018-08-20:
Allright that clears it up (will close issue). Thank!

closed 2018-08-20 by jochemverboom