realsenseai/librealsense #2277: rs2_pipeline_start config error

arikyueh, 2018-08-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.09.13.00)| 
| Operating System & Version |   Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  (4.4.38-tegra)                                         | 
| Platform                                 | Jetson TX2 |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++/opencv}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<Trying to implement left and right IR images for stereo vision. I am implementing the same code as the thread in #1140. 

	int width = 480;
	int height = 270;
	int fps = 6;
	rs2::config config;
	
	config.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
	config.enable_stream(RS2_STREAM_INFRARED, 2, width, height, RS2_FORMAT_Y8, fps);

	// start pipeline
	rs2::pipeline pipeline;
	rs2::pipeline_profile pipeline_profile = pipeline.start(config);

Running it I encounter an error on the line with pipeline.start(config);

RealSense error calling rs2_pipeline_start_with_config(pipe:0x51cbf0, config:0x4675f0):
    Failed to resolve request. No device found that satisfies all requirements

Looking into the error I tried commenting out one of IR cameras and the pipeline worked. Error happens specifically when I have left and right IR cameras in my config.

>

ywjia, 2018-09-08:
I experience the same issue, on Windows 10, D435, SDK 2.0.
With the following code:

rs2::context ctx;
auto list = ctx.query_devices();
if (list.size() == 0) 
    	throw std::runtime_error("No device detected. Is it plugged in?");
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 15);
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 15);
rs2::pipeline_profile profile = pipe.start(cfg);

I got the following error message: 

"RealSense error calling rs2_pipeline_start_with_config(pipe:000001F6332C9860, config:000001F6332C9620): 
Couldn't resolve requests."

The same code works fine in Ubuntu 16.10.

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
Hi arikyueh,

We tested your configuration off SDK 2.15, by using rs-capture example and modifying to be with your settings.
There is NO issue.

Can you please test it with RealSense Viewer and configure to be streaming 2 infrared streams w/ the configuration, and see if there is any issue there?

Thanks!

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
Hi ywjia,

Guess you missed some digits in the SDK information that you posted here.
Not sure which SDK version you used.

Can you please download the same version of RealSense View executable file from the github, and test it out with the same configuration with the viewer?
I don't have issue here with various RealSenseViewer versions.

Also can you try to separate depth and color, and test one by one to see which causes the issue? Please let us know.

Thanks!

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
Hi ywjia,
Thanks for the update. Glad to hear that the problem is solved after your upgrade.

Thanks!

RealSense-Customer-Engineering, 2018-09-17:
[Realsense Customer Engineering Team Comment]
Hi arikyueh,

Do you still need help on this one?

Thanks!

closed 2018-09-24 by RealSense-Customer-Engineering

HowyZax, 2019-12-27:
It should be noted that I had the camera or driver get into a bad state which cased an unknown exception to be thrown every time I called pipe.start. The RealSense viewer would freeze when trying to connect to the camera in this state.  I re-started (not power cycled) windows, pulled the USB cable out of the camera several times and could not get it to work. The solution was pulling the USB cable out of the computer itself (as strange as that may seem). I suspect a real power cycle on the PC would have fixed it too. I have seen other instances where a dangling USB cable caused problems on other devices.