realsenseai/librealsense #2280: RuntimeError: null pointer passed for argument "frame_ref"

KeCh96, 2018-08-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |  4.15.0 | 
| Platform                                 | Linux |
| SDK Version                            |  SDK2  | 

I try to read depth image by the following code:
```
pipeline = rs.pipeline()
pipeline.start()
while True:
    frames = pipeline.wait_for_frames()
    depth  = frames.get_depth_frame()
    depth_data = depth.as_frame().get_data()
    np_image = np.asanyarray(depth_data)
```
But I run into an error:
```
Traceback (most recent call last):
  File "demo_2d_6class_mirror.py", line 374, in <module>
    depth_data = depth.as_frame().get_data()
RuntimeError: null pointer passed for argument "frame_ref"
```

Could you please help me? @dorodnic @zivsha @RealSense-Customer-Engineering

ev-mp, 2018-08-21:
Hello @KeCh96 ,
The code doesn't check the return value to establish whether the depth frame is valid.
Please see the [python example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L21) reference

KeCh96, 2018-08-23:
Hello, @ev-mp .
I tried `if not depth: continue`, but the error still occurs.
Note that even when I plug out the camera, the code also run into `RuntimeError: null pointer passed for argument "frame_ref"`.

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @KeCh96 ,

I can't reproduce this issue.
Please try again with new version librealsense.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@KeCh96 Did you get chance to try the latest librealsense? Any good news? Looking forward to your update. Thanks!

drzraf, 2018-10-31:
In my case I was using 2.14.1
Will try soon with latest...

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Thank you, @drzraf . You are right.
@KeCh96, please `enable_stream` in config object before pipeline.start, and call pipeline.start with param config.

closed 2018-11-20 by RealSense-Customer-Engineering