realsenseai/librealsense #2281: Some questions on HW synchronization

vladsterz, 2018-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC.  |
| SDK Version                            | 2.14.0                          | 
| Language                            |  C++                         | 

### Issue Description
For the days I've been trying to use HW sync to (of course) synchronize 2 D415 devices, either remotely or on the same machine and my results have not been exactly robust. So, some questions have arose on how to setup properly a HW synchronized system, beyond the suggestions of this [whitepaper](https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf)

- **RC filter**
How crucial is it to have filters at the input of the sensors. When do ESD events happen and what do they depend on?

- **Frame Drop**
Suppose a system of two devices, a master and a slave.
  - What **should** happen when master drops a frame.
  - When master drops a frame, does he produce signal at his output pins?If not, will slave device still acquire frame?
  - Can the slave device fail to acquire a frame despite the fact master did acquire frame?

- **C++ API frame acquizition**
What is the _preferred_ way of acquiring frames in a HW synced system (wait4frames,poll4frames,async_syncer,syncer)? Does it even matter?

-  **Inter Camera Synchronization Setting**
   - As I have observed ,after having set the sensors to be master/slave 
`depth_sensor1.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1);`
`depth_sensor2.set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 2);`
a restart of the devices (running application) is necessary. Does this observation holds?Is it something anyone else has observed?
   - In the case that a restart is needed (as previously stated), sometimes devices fail to acquire and frames and a unplug/plug is needed. Again, anyone has observed this behavior? Is there a better way overcoming this problem? Unplugging/re-plugging resets the setting master/slave, so this leads to question 1 of this section :P .

- **Exposure Duration**
I've seen somewhere in an issue that in order to have HW sync, one must disable Auto Exposure and set it manual at some levels that I fail to find. What are those values? Do I have to disable AE in both sensors (color - depth)? Let's say that depth AE must be set to **X**, does in have to be exactly this value or I can go a little higher or lower?How does it affect the HW sync functionality? 

Thanks in advace, Vlad.

vladsterz, 2018-08-24:
### Update
Most of the times, setting master and slave configurations, would result in the _slave_ camera not being able to acquire frames. After messing around (a lot) with my code I think I got a kind of stable and robust configuration. 
**Steps**
_for_every_device_do_
{
 - hardware_reset() & sleep(3)
 - set_master (or slave,w/e is needed) , set_autoexposure(OFF) (both color and depth) , set_exposure_times(depth=33000,color=315)
 - open_sensors() (color and depth)
 - start_streaming() (color and depth)
 - sleep(2)

}

**Notes**
 - Acquisition method is _rs2::sync_
 - Set the first device as _master_

RealSense-Customer-Engineering, 2018-09-12:
[Realsense Customer Engineering Team Comment]
Hi @vladsterz,

That's great that you found your way to resolve. More info as follows:
In summary of HW sync,
1. Need to first configure the cameras to either master or slave mode and switch sensors to manual exposure and the same fps (preferred) . THEN start the stream on the master camera first before the slaves.
2. The ESD protection is to improve the signal integrity and false trigger on the slave camera. When it happens the false trigger could cause the RGB sensor (sometimes depth sensor) on the slave camera(s) to stop streaming.
3. The hw sync trigger happens at the imaging sensor level. One camera drops frame should not affect another, regardless of hw-sync or non-hw-sync.

closed 2018-09-26 by RealSense-Customer-Engineering