realsenseai/librealsense #2287: SR300 connection problem

lukashermann, 2018-08-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |SR300 | 
| Firmware Version                   | Unknown | 
| Operating System & Version |   Ubuntu 16.04 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-32-generic                                        | 

I'm trying to connect the camera with my computer, but it is not detected by the realsense-viewer. If I run `lsusb`, it shows `Bus 003 Device 037: ID 041e:409b Creative Technology, Ltd` instead of the required intel device. 
The output of `dmesg` shows 

```
[ 5578.057327] usb 3-1.1.2: Product: BlasterX Senz3D VF0810
[ 5578.057328] usb 3-1.1.2: Manufacturer: Creative Technology Ltd
```
I couldn't find information about if there is a difference between the **INTEL SR300** and the **Creative BlasterX Senz3D**, but at least optically they look the same. Can you provide me with information about how to make realsense recognize my camera?

**Additional information:**
I installed librealsense2 from source with the required kernel patches and no problem occurred during the installation. The cam is connected via a USB 3 hub with power supply that itself is also connected to a USB 3 port.

dorodnic, 2018-08-21:
Hi @lukashermann 
At least at first glance everything should be working. 
What about `lsusb | grep 8086`? Any errors when you run `rs-enumerate-devices`?

lukashermann, 2018-08-23:
I noticed that the kernel modules that are installed in the ubuntu kernel patch are not loaded by default after a reboot. This is the output of `lsmod`:
```
Module                  Size  Used by
rpcsec_gss_krb5        40960  0
nfsv4                 577536  5
nfs                   262144  5 nfsv4
fscache                65536  2 nfsv4,nfs
binfmt_misc            20480  1
joydev                 24576  0
snd_usb_audio         196608  1
snd_usbmidi_lib        32768  1 snd_usb_audio
input_leds             16384  0
snd_hda_codec_hdmi     49152  1
snd_hda_codec_realtek    98304  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
snd_hda_intel          40960  5
snd_hda_codec         126976  4 snd_hda_intel,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_realtek
snd_hda_core           81920  5 snd_hda_intel,snd_hda_codec,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_realtek
snd_hwdep              20480  2 snd_hda_codec,snd_usb_audio
snd_pcm                98304  5 snd_hda_intel,snd_hda_codec,snd_usb_audio,snd_hda_core,snd_hda_codec_hdmi
snd_seq_midi           16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  2 snd_seq_midi,snd_usbmidi_lib
snd_seq                65536  2 snd_seq_midi_event,snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
snd_timer              32768  2 snd_seq,snd_pcm
snd                    81920  25 snd_hda_intel,snd_hwdep,snd_seq,snd_hda_codec,snd_usb_audio,snd_timer,snd_rawmidi,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_usbmidi_lib,snd_seq_device,snd_hda_codec_realtek,snd_pcm
mei_me                 40960  0
mei                    90112  1 mei_me
soundcore              16384  1 snd
lpc_ich                24576  0
shpchp                 36864  0
eeepc_wmi              16384  0
asus_wmi               28672  1 eeepc_wmi
sparse_keymap          16384  1 asus_wmi
video                  45056  1 asus_wmi
wmi_bmof               16384  0
nvidia_uvm            671744  0
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
coretemp               16384  0
kvm_intel             212992  0
kvm                   598016  1 kvm_intel
irqbypass              16384  1 kvm
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
pcbc                   16384  0
aesni_intel           188416  0
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
glue_helper            16384  1 aesni_intel
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel,aesni_intel
intel_cstate           20480  0
intel_rapl_perf        16384  0
wmi                    24576  2 asus_wmi,wmi_bmof
mac_hid                16384  0
nfsd                  339968  13
auth_rpcgss            61440  2 nfsd,rpcsec_gss_krb5
nfs_acl                16384  1 nfsd
lockd                  94208  2 nfsd,nfs
grace                  16384  2 nfsd,lockd
sunrpc                335872  43 auth_rpcgss,nfsd,rpcsec_gss_krb5,nfsv4,nfs_acl,lockd,nfs
parport_pc             36864  0
ppdev                  20480  0
lp                     20480  0
parport                49152  3 lp,parport_pc,ppdev
autofs4                40960  4
nvidia_drm             45056  1
[... 39 lines cut ...]
[ 1024.553142] usb 4-1.4.4-port1: attempt power cycle
[ 1024.949388] usb 4-1.4.4.1: new SuperSpeed USB device number 17 using xhci_hcd
[ 1025.157077] usb 4-1.4.4.1: unable to read config index 0 descriptor/all
[ 1025.157083] usb 4-1.4.4.1: can't read configurations, error -71
[ 1025.489473] usb 4-1.4.4.1: new SuperSpeed USB device number 18 using xhci_hcd
[ 1025.700949] usb 4-1.4.4.1: unable to read config index 0 descriptor/all
[ 1025.700956] usb 4-1.4.4.1: can't read configurations, error -71
[ 1025.701211] usb 4-1.4.4-port1: unable to enumerate USB device
[ 1083.749725] usb 3-1: USB disconnect, device number 10
[ 1083.749726] usb 3-1.4: USB disconnect, device number 11
[ 1083.749727] usb 3-1.4.4: USB disconnect, device number 12
[ 1083.749729] usb 3-1.4.4.2: USB disconnect, device number 13
[ 1083.769916] usb 4-1: USB disconnect, device number 12
[ 1083.769917] usb 4-1.4: USB disconnect, device number 13
[ 1083.769918] usb 4-1.4.4: USB disconnect, device number 14
[ 1087.506322] usb 3-1: new high-speed USB device number 14 using xhci_hcd
```

The errors like
```
[ 1024.011924] usb 4-1.4.4.1: unable to read config index 0 descriptor/all
[ 1024.011931] usb 4-1.4.4.1: can't read configurations, error -71
```
didn't always occur and I don't know what causes them, but it any case the camera is not detected.
I hope the additional information helps.

RealSense-Customer-Engineering, 2018-09-05:
[Realsense Customer Engineering Team Comment]
Hi @lukashermann,

Per your log, only Creative USB VID/PID is found, don't see SR300 device is enumerated correctly. Since the USB enumeration phase, its config can't be read normally. To further clarify:
1. Can you try to connect to the USB port directly instead of USB3 hub?
2. Did you try another PC or different OS like Windows to see if SR300 device can show correctly?

[1023.877499] usb 3-1.4.4.2: New USB device found, idVendor=041e, idProduct=409b
[1023.877503] usb 3-1.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1023.877505] usb 3-1.4.4.2: Product: BlasterX Senz3D VF0810

[1024.345378] usb 4-1.4.4.1: new SuperSpeed USB device number 16 using xhci_hcd
[1024.552899] usb 4-1.4.4.1: unable to read config index 0 descriptor/all
[1024.552906] usb 4-1.4.4.1: can't read configurations, error -71

lukashermann, 2018-09-07:
Yes I also tried to connect it directly, but this didn't change anything. It works on another Linux machine without problems.

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @lukashermann,

You need to verify that uvcvideo is enabled in the kernel configuration - try to run "`modinfo uvcvideo | grep version "`
 to obtain driver's info.
If the uvc driver is available but fails to load automatically, you can try to run "sudo apt-get install --reinstall linux-image-generic linux-image" to reset the modules.

And, how about your SR300 firmware version? You can try to upgrade to 3.21.0 and see how it works.

RealSense-Customer-Engineering, 2018-09-19:
[Realsense Customer Engineering Team Comment]
Hi @lukashermann,

any update?

lukashermann, 2018-09-24:
Uvcvideo was enabled in the kernel configuration after I manually loaded it with `sudo modprobe uvcvideo`. I spoke to my coworker who said he had similar problems trying to use a Kinect camera with the same computer, his suggestion was that it is an issue with the motherboard's USB controller. So I switched to another machine where it is working now without flaws. Thank you for your help and patience!

closed 2018-09-24 by lukashermann