realsenseai/librealsense #2288: rs-post-processing problem

temmat, 2018-08-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (5.9.14.0) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |  2.15.0                           | 
| Language                            |  C++/ VS2017                  | 
| Segment			|  others                          | 

### Issue Description

The rs-post-processing.cpp example doesn't work.

Firstly the ImGui used in librealsense seems to be quite old (1.49 to be precise), the init function calls have been changed from that, but using the ImGui directly from the librealsense source "third-party" folder is ok for this (building with Cmake does this).

What I'm actually having problems with is in VS2017 with a "no suitable conversion function from "window*" to "GLFWwindow*" in the init function call:
ImGui_ImplGlfw_Init(app, false);

From what I understand a GLFWwindow is window+context, so ... should the window declaration actuall be done differently than in the example or is my VS2017 setup missing something else ?

Thank you for any help / hint.

temmat, 2018-08-27:
For anyone interested in this, the rs-measure example has an implementation of the post processing filters (which works for windows too). It's a different approach though (than the example in the source files) and the hole-filling filter is not implemented, but still, you can see how it works I guess.

If you want to add the hole filling filter yourself, then all you need to do is declare it at the beginning of the main function and then call it in the video_processing_thread after the disparity2depth transformation.

With this I don't consider the issue solved (because the actual rs-post-processing example is still missing in the windows build), but developers can still get some idea how to do it from this other example.

temmat, 2018-09-02:
Thank you for the comment an your attempt to get the same error.

To clarify: I was talking about the prebuilt Windows sdk version 2.15.0.163. which does not include the post-processing example. I tried installing the sdk again, and the said example is simply not there. I haven't built the sdk from source yet, but it seems I might have to to get that specific example working. Or is it perhaps included in the newer 2.16 build ?

(I got the error "no suitable conversion" when I quickly tried to compile just the post-processing example from https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp   with VS2017 without using Cmake, )

dorodnic, 2018-09-03:
Sorry, I stand corrected. I checked that all the installer examples compile cleanly using VS17, but post-processing is not in the installer. 
So far we shied away from redistributing too much 3rd-party libraries in the installer (this is the reason for both post-processing and align missing - both would pull IMGUI and GLFW into the installer) We will revisit this decision since this is not the first issue bringing it up.

dorodnic, 2018-09-03:
Here are my concerns with adding GLFW and IMGUI (and similarly OpenCV) into the installer - 
1. librealsense API is designed to allow you to compile the library with one version of Visual Studio and the application with another. This is not true for libraries in general, and there are plenty of examples that require a dedicated DLL for every version of Visual Studio (PCL for instance). I think GLFW and IMGUI are also Visual Studio agnostic, but we need to check.
2. Every binary we distribute has Intel digital signature to let users know they are using an official version. It would be a bit odd for us to distribute 3rd-party libraries in this fashion.

temmat, 2018-09-03:
I thought glfw was already included, since there are some glfw callbacks and structures named accordingly within librealsense (at least judging by the comments) in the rs-measure example ( specifically "register_glfw_callbacks") and in samples\example.hpp (specifically the glfw_state structure). But maybe those are just names and have actually no dependency on the glfw library.

I understand your concern with including glfw or imgui (or any other third party library) with librealsense - it's objectively "cleaner" not to, and of course represent more dependencies to deal with if expanding the sample code to something else. Nevertheless the examples that you mention yourself (post-processing and align) show/propose how to deal with librealsense in a highly relevant matter, so I propose to either include the examples and the necessary libraries in the upcoming releases OR (as an alternative solution) use another simple method to provide an input interface needed for the manipulation of example parameters ... I guess even a very basic console input/text in a separate window could produce a similar result ?

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@temmat Did you still have issue of rs-post-processing? Please update. Thanks!

temmat, 2018-10-22:
Well, both post-processing and align samples are still not part of the SDK installer and there is no replacement unless the user builds the samples themselves from github.

For me personally it is solved, it is however not for any other user using the SDK installer.

dorodnic, 2018-10-24:
This issue should be resolved in the upcoming 2.16.2 release. The installer is going to include the missing examples and redistribute GLFW, IMGUI and OpenCV. Once the release is ready, I hope it will help new users. @temmat thank you for the valuable feedback.

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
@temmat Could you please have a try with 2.16.2 to see if you still get the problem? Thanks!

HippoEug, 2018-11-06:
I had the same issue written [here](https://github.com/IntelRealSense/librealsense/issues/2486) as well.

I will test the latest release version when it is out, since 2.16.2 and 2.16.3 are still pre-releases!

RealSense-Customer-Engineering, 2018-12-10:
[Realsense Customer Engineering Team Comment]
@temmat @HippoEug Any good news? Please update. Thanks!

HippoEug, 2018-12-13:
Yeah it is OK for me !

RealSense-Customer-Engineering, 2018-12-13:
[Realsense Customer Engineering Team Comment]
@HippoEug That's great.
@temmat How about your test?

temmat, 2018-12-17:
Align and post-processing are included and build for me too. Tested with latest version 2.17.0.394 windows SDK installer.

closed 2018-12-17 by temmat