realsenseai/librealsense #2289: Long distance range settings

jgomezlopez, 2018-08-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415/D435                                            | 
| Operating System & Version |  Linux (Ubuntu 16                                  | 
| Kernel Version (Linux Only)    |  4.14.13                                                  | 
| Platform                                 | x86_64                                                    |
| SDK Version                            |  2.15                                                      | 
| Language                            |  C++                                                          | 

Hello,

I would like to know which are the proper settings for measuring accurate distances between 3.5m and 6m. I played with Depth Quality tool without success. In that range I get variations for the same surface higher than 40 cm.

If you can also tell me some tips about how to tune the settings depending on the distance range is fine.

Thanks  in advance,

Jose Gomez

MartyG-RealSense, 2018-08-22:
Pages 9 and 13 to 16 of the beautifully illustrated camera tuning guide give tips on accuracy over distance.  The guide as a whole is well worth reading too.

https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf

MartyG-RealSense, 2018-08-22:
Page 13 of the guide has a formula for calculating RMS error.

[image]

As the RMS / Depth chart on that page suggests, a high amount of RMS error at distances up to 6 m may be unavoidable if using the D435 camera model.  The D415 has less RMS error over distance.

Rather than use disparity shift to change MinZ and MaxZ, I would be more inclined in this case to change the depth units scale, as described on page 21.

[image]

RealSense-Customer-Engineering, 2018-09-17:
[Realsense Customer Engineering Team Comment]
Hi jgomezlopez,

Do you still need help on this one?

Thanks!

closed 2018-09-24 by RealSense-Customer-Engineering