realsenseai/librealsense #2290: Multiple D400 with NUC 7

holstenholsten, 2018-08-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Win 10 |                                
| Platform                                 | NUC |

Hi! I was wondering if anyone has any experience with multiple D400's connected to a single NUC 7?
https://click.intel.com/ready-to-go-intelr-realsensetm-nuctm-complete-kit-boxnuc7i3bnhxf-d415.html

I have 3 D415. If I buy the bundle, can I then connect 4 cams to the NUC?

Thanks!

AntoineWefit, 2018-08-22:
I think this setup will work, but it is more difficult to predict its performances. I'd say it'll be conditioned by your application, for example, streaming + rendering is quite resources consuming, particularly at high resolutions. However, if you plan to do "more or less" instantaneous capture (images or points clouds), I don't think it'll be a problem.
From what I know, your main concern when selecting a board is having the same amount of USB 3.0 ports and cameras. Those are required to achieve max performances (res/fps).

If you haven't already, you should take a look at [this paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf), it may answer some of your questions.
You also might be interested by [this post](https://communities.intel.com/thread/127437).

If you have the opportunity to actually test the setup, please let me know because I'm also interested.

MartyG-RealSense, 2018-08-22:
Thanks guys for the answers above.  

I would add that according the store listing for the NUC 7, it has 4 external USB 3.0 ports (2 front, 2 back).  So it should certainly be viable to connect 4 cameras to it.  The 2 USB 2.0 ports are listed as internal though, so it might be awkward to connect other peripherals such as keyboard and mouse if you have used up all the external ports with cameras.

You could free up some ports by using a USB hub.  Attaching the cameras directly to the computer's ports is the preferred approach though, as it should help to ensure that each camera has its own independent USB controller, whereas on a hub the devices may share a controller.

The link below has advice about using the internal USB of NUCs.

https://communities.intel.com/message/502355#502355

holstenholsten, 2018-08-22:
Thank you so much for your help!
I will get back to you if we go for it.

MartyG-RealSense, 2018-08-22:
You're very welcome!   :)

Intel announced a new range of budget NUCs this week that may also suit your needs.

https://communities.intel.com/thread/128237

At the high end of the budget range, this year's super-powerful NUC 8 VR model has 7 USB 3.0 ports (4 rear, one front-charging and 2 internal).

https://communities.intel.com/message/534707#534707

closed 2018-09-20 by RealSense-Customer-Engineering