realsenseai/librealsense #2292: Viewer cannot stream two D415 devices

hflsdupont, 2018-08-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.15                          | 
| Language                            |  C++                         | 

### Issue Description
I'm using the latest version of Viewer (2.15.0). It works OK when I only start one D415 device. As I stream two devices, the live videos are not playing simultaneously, always having streams for one camera be stopped. The issue does not happen on two D435 cameras.

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hello @hflsdupont,

Thank you for pointing out this issue!  We were able to replicate the issue with RealSense Viewer and will work to create a fix shortly.
Please stay tuned, will update this issue when fix is implemented.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hello @hflsdupont,

Apologies for the delay, this issue should now be resolved with LRS 2.16.2, and newer LRS 2.16.3.

-  LRS 2.16.2 and 2.16.3 are now both available on public github.
  >  2.16.2 (https://github.com/IntelRealSense/librealsense/tree/v2.16.2)
  >  2.16.3 (https://github.com/IntelRealSense/librealsense/tree/v2.16.3)

-  Ran tests with RealSense Viewer 2.16.2 / 2.16.3 on Win 10 RS4 / Ubuntu 16.04, attached multiple D415/D435 cameras with FW 5.10.6 (https://downloadcenter.intel.com/download/28237/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t).

Able to confirm that cameras now stream simultaneously with RealSense-Viewer.

Please let us know if problem is fixed with either of the two newer versions of LibRealSense (see above links).

If no response in 5 days, will close this ticket as solved.

Thank you!

RealSense-Customer-Engineering, 2018-11-06:
[Realsense Customer Engineering Team Comment]
Close this issue as solved as there has been no response on this ticket.  Also fix is validated with LRS 2.16.2 and newer.

Thank you.

closed 2018-11-06 by RealSense-Customer-Engineering