realsenseai/librealsense #2295: Frame Synchronization d400

rafaor29, 2018-08-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                         | 
| SDK Version | 2.14.0 |
| Platform                                 | PC |

### Issue Description

Hi, i've been working with the multicamera example so it is able to stablish the hardware synchronization of the connected cameras (in this case 2 d415) and it also captures an image like in the example of save-from disk, each camera in a different thread. 

I've made some captures and the metadata retrieved from the captures shows that the frame counters are almost always different, from both cameras and also in the same camera color and depth are still different. I don't understand exactly why this is happening, both hardware and software synchronization work in other test applied before.

Also, when we did test with a chrono the difference between the time in each picture taken by the cameras was 0.05 seconds.

I'll attach the excel with relevant data from all the metadata of the test.
[Muestras_415.xlsx](https://github.com/IntelRealSense/librealsense/files/2313739/Muestras_415.xlsx)

Thanks for the help.

MartyG-RealSense, 2018-08-23:
When you are doing hardware sync between your two D415s, may I please confirm that you have connected the two cameras together, as described in the multi-cam sync white paper document linked to below?

[image]

https://realsense.intel.com/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper_rev1.1.pdf

Also, have you set one of the D415s as a Master and the other as a Slave?  

RealSense Group CTO Anders said recently: "When you set a camera to slave it will still acquire frames at the right frame rate even if there is no sync input signal.  It tries to sync but if there is no signal that comes in at the right time, it gives up and triggers itself."

rafaor29, 2018-08-23:
Yes, the cameras are connected following the instructions of the white paper. 
Also, the master mode is stablish like this:
'rs2_set_option((const rs2_options*)sen, RS2_OPTION_INTER_CAM_SYNC_MODE, 1, e);'
and the slave mode: 'rs2_set_option((const rs2_options*)sen, RS2_OPTION_INTER_CAM_SYNC_MODE, 2, e);'
Also i've checked that there is an external sync pulse so the cameras should be master and slave respectively. Is it possible that even with the pulse generated the cameras are still not synchronized?

MartyG-RealSense, 2018-08-23:
Sync user @vladsterz also had questions on the subject a couple of days ago.   You may find something useful in their own post if you haven't seen it already.

https://github.com/IntelRealSense/librealsense/issues/2281

vladsterz, 2018-08-24:
@rafaor29  Hey. I can confirm that frames counters are different in the two devices, which makes a lot of sense in my opinion. Since _master_ and _slave_ devices can start independently (or even if _master_ has to start first, the statement holds), they will start acquiring frames at different times. Let's suppose that they start at the very exact moment. Now if _slave_ drops a frame, different counters will occur once again. I am not sure what happens with the trigger pulse when _master_ drops a frame, but it will lead to different counters as well.

rafaor29, 2018-08-24:
Hi @vladsterz . I've seen that we're experiencing similar issue. I agree that it has sense that the frame counter is different so you can align the frames later. Still, what i think it is happening that the slave camera is not getting the signal at the right time as said before in this post, and it is triggering itself. 

After stablishing the master and slave devices, how do you check each one is the one it is supposed to be?

Also, what you said, in the excel that i uploades you can find 2 cases in which a frame drop occurs in camera 74 and the difference between the timestamps of the color and depth sensors is bigger than usual.

vladsterz, 2018-08-27:
Hey @rafaor29 . It's is easy to check whether a device is master or slave. 
`depth_sensor.get_option(RS2_OPTION_INTER_CAM_SYNC_MODE)` is the command.
I have not checked the color-depth sensor sync, since I want to establish inter-device syncing at first.

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
As for RGB-depth HW sync, please see #774.

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
Hi @rafaor29,

Any update? Still need any support for this topic?

RealSense-Customer-Engineering, 2018-11-16:
[Realsense Customer Engineering Team Comment]
Please reopen if you require more information

closed 2018-11-16 by RealSense-Customer-Engineering