realsenseai/librealsense #2299: Illegal instruction on non-AVX processor

kjkjava, 2018-08-23:
-------------------------------------------------------------------------------
| Required Info                  |                                                   |
|---------------------------------|------------------------------------------- |
| Camera Model                        | Any |
| Firmware Version                    | Any |
| Operating System & Version   | Ubuntu 14.04 | 
| Kernel Version (Linux Only)    | 4.4.0 | 
| Platform                                   | Pentium |
| SDK Version                            | 2.14.1 |
| Language                                | C++ |

### Issue Description
As mentioned in previous issues (#1428 #1443, #1464, #1484), there have been issues executing AVX instructions on CPUs that don't support AVX (causing SIGILL, "Illegal instruction (core dumped)"). Using SDK version 2.14.1, the issue is two-fold:

1. If I compile with `-mavx`, my application crashes at global initialization. I see that there is auto-selection code to bypass the AVX implementation if not supported on a CPU, but in debugging, I saw crashes in two different locations (one in librealsense and one in another library). Generally speaking, adding `-mavx` gives the compiler free-reign to optimize using AVX wherever it wishes, so I don't think this should always be enabled for non-ARM builds.

2. If I remove `-mavx`, the build fails due to SSSE3 guards (which my processor supports) where AVX guards should be used instead. I have a patch for this, which is ultimately the solution to my issue. I'll send a pull request with this patch shortly.

dorodnic, 2018-09-06:
Thanks @kjkjava 
We compile with `-maxv` only `image_avx.cpp` ([CMakeLists.txt:849](https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt#L849))
So with your PR functions implemented in `image_avx` will never get called. 
I don't think that the compiler will auto-generate AVX unless we explicitly allow it (after all machines without AVX are not that uncommon)
Do you think we are still missing some case?

dorodnic, 2018-09-25:
I think we should be good now, but feel free to open a new issue / PR if you identify something.
Thank you!

closed 2018-09-25 by dorodnic

kjkjava, 2018-09-25:
Hey @dorodnic, sounds reasonable, especially since it's just image_avx.cpp. I think we had just enabled it for the entire build as a workaround for the failed build fixed in this bug.