realsenseai/librealsense #2304: Get images from my own computer, instead of RealSense

KeCh96, 2018-08-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |  4.15.0 | 
| Platform                                 | Linux |
| SDK Version                            |  SDK2  | 

I try to read image by the following code:
```
pipeline = rs.pipeline()
pipeline.start()
while True:
    frames = pipeline.wait_for_frames()
    depth  = frames.get_depth_frame()
    depth_data = depth.as_frame().get_data()
    np_image = np.asanyarray(depth_data)
```
I found when I use `frames.get_depth_frame()`, the realsense camera do not open, instead the camera of my computer opens. Then I get error: `RuntimeError: null pointer passed for argument "frame_ref"`
Similar situation happens when I use `frames.get_depth_frame()`, the realsense camera do not open. Unexpectedly, I get image from my own computer camera. 
Does anyone know the solution? Thanks in advance. @dorodnic @zivsha

AntoineWefit, 2018-08-24:
I don't really know how the SR300 camera behave, but I had a similar issue with a D4XX. I'm guessing that you are not specifying precisely enough which device you want to access, and the first one might then be selected by default (which can be your webcam). Maybe try to display a list of the connected devices to understand which one is which.
Here is an idea of the principle (work for D4XX coded in C++) : 
```
// Create a context
rs2::context cxt;
// Get all devices connected
auto device = cxt.query_devices();
// Get number of devices
size_t device_count = device.size();
// Select the first device
auto dev = device[0];
// Display the list of connected devices
for (auto i = 0; i < device_count; i++)
{
	auto hardware = device[i];
	cout << "Name of the camera " << i << ": " << hardware.get_info(RS2_CAMERA_INFO_NAME) << endl;
}
```

vladsterz, 2018-08-24:
@KeCh96  Haven't tried the Python wrapper yet, but there are examples [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python). Always check if you have acquired data (like [here](https://github.com/IntelRealSense/librealsense/blob/65ce72a59bcc80f95ed0150c36573077ce4311a2/wrappers/python/examples/opencv_viewer_example.py#L28)).

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
Hi KeCh96,

Wonder if you tried what were suggested by AntoineWefit and vladsterz?
If there is nothing else, we are going to close this one.

Thanks!