realsenseai/librealsense #2306: Regression in v2.15.0: color stream shown blank in realsense viewer expect for YUYV

staniliev, 2018-08-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Ubuntu Mate  | 
| Kernel Version (Linux Only)    | Linux  4.14.55                                         | 
| Platform                                 | odroid XU4  |
| SDK Version                            | build from source librealsense v2.15.0                          | 
| Language                            | C                          | 
| Segment			|  others                           | 

### Issue Description
I open realsense viewer and start color stream in RGB8, which results in a blank screen.
Depth, infrared streams and color stream in YUYV format work fine. The only color stream format that works is YUYV.
When I sync and build librealsense with "tags/v2.14.1" everything works correct (all color stream formats work). When I sync and build with "tags/v2.15.0" I get blank output for RGB8 color stream.
My kernel is patched and "hardware clock" is shown when I click on "info" in the depth stream window.

This is the console output - the last line with "Non-zero status (-71) in video completion handler." looks fishy:

[  208.314353] usb 4-1.2: new SuperSpeed USB device number 3 using xhci-hcd
[  208.335144] usb 4-1.2: New USB device found, idVendor=8086, idProduct=0ad3
[  208.335150] usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  208.335155] usb 4-1.2: Product: Intel(R) RealSense(TM) 415
[  208.335160] usb 4-1.2: Manufacturer: Intel(R) RealSense(TM) 415
[  208.335165] usb 4-1.2: SerialNumber: 812113024034
[  208.368715] videobuf2_vmalloc: loading out-of-tree module taints kernel.
[  208.374167] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad3)
[  208.376065] uvcvideo: Unable to create debugfs 4-3 directory.
[  208.376250] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[  208.376258] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[  208.376266] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) 415 RGB was not initialized!
[  208.376272] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[  208.376500] input: Intel(R) RealSense(TM) 415: Int as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input6
[  208.376898] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 415 (8086:0ad3)
[  208.377249] uvcvideo: Unable to create debugfs 4-3 directory.
[  208.377369] uvcvideo 4-1.2:1.3: Entity type for entity Processing 7 was not initialized!
[  208.377376] uvcvideo 4-1.2:1.3: Entity type for entity Extension 8 was not initialized!
[  208.377382] uvcvideo 4-1.2:1.3: Entity type for entity Camera 6 was not initialized!
[  208.377592] usbcore: registered new interface driver uvcvideo
[  208.377596] USB Video Class driver (1.1.1)
[  380.462643] uvcvideo: Non-zero status (-71) in video completion handler.

neuralassembly, 2018-09-06:
|Required Info||
| ---- | ---- |
|Camera Model|D415 &amp; D435|
|Firmware Version|5.10.3|
|Operating System &amp; Version|Ubuntu MATE 16.04.5|
|Kernel Version (Linux Only)|Linux 4.14.17|
|Platform|Raspberry Pi 3|
|SDK Version|build from source librealsense v2.16.0|
|Language|C|
|Segment|others|

Same situation. With v2.16.0, the color stream works only in YUYV format.
With v2.14.0 the color stream worked fine in all the formats.

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
Hello @staniliev,

Thank you for pointing out this issue! We were able to replicate the issue with RealSense Viewer in Ubuntu MATE and will work to create a fix shortly.
 Please stay tuned, will update this issue when fix is implemented.

dorodnic, 2018-09-12:
Yes, seems to be affecting all non-x86, non-Android platforms. 
IMHO, switching between the two lines [src/image.cpp:312-313](https://github.com/IntelRealSense/librealsense/blob/master/src/image.cpp#L312-L313) solves the problem.

jamesbellaero, 2018-09-17:
The tabbing is off in that file actually. What you need to do is tab over the #ifdef __AVX2__ block and add an extra #endif after line 606. This fixed it for me.

jamesbellaero, 2018-09-19:
@dorodnic Would you like me to fork it, fix the bug, and make a pull request?

dorodnic, 2018-09-25:
Should be resolved in [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
@dorodnic

Able to confirm v2.16.1 resolves color stream issue in previous versions of LRS.

We will close this issue now.
Thank you.

closed 2018-09-26 by RealSense-Customer-Engineering