realsenseai/librealsense #2307: Unity crashes when playback recorded file

frenchtouchXR, 2018-08-24:
Hello,

I work on Unity/PC with the latest version of the wrapper.

When I record a file (record mode) and then playitback. Unity crashes. 

It didn't happen with previous versions. 

What do you think about it ? 

Best regards

frenchtouchXR, 2018-09-08:
I updated to the 2.16 version and upgraded the RS firmwares to 5.10.3.0. I can record but the playback crashes everytime.

frenchtouchXR, 2018-09-11:
Please help :(

RealSense-Customer-Engineering, 2018-09-14:
[Realsense Customer Engineering Team Comment]
Hi @frenchtouchXR,

This issue can be reproduced. Will further check internally.

frenchtouchXR, 2018-09-24:
Hi, 

Any news of a patch so far ? 

Best

velenoise, 2018-09-28:
+1 for that, having the same issue. Unity 2018.1.6f1

frenchtouchXR, 2018-09-29:
I tried again with the 2018.1.6f1 version. The problem is still the same. My researches on volumetric capture stopped 2 ago.

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
@matkatz,
Do you have any update about this issue?

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
Hi @velenoise,

What's your unity/firmware versions and test steps? record and playback ros bag file both in Unity? I didn't see this issue on 2.14.

matkatz, 2018-10-02:
This issue can be reproduced, we will provide a fix soon.

velenoise, 2018-10-02:
@RealSense-Customer-Engineering 

I'm on Unity 2018.1.6f1, using firmware 05.10.03.00. I had tested previously with a recording made on Realsense Viewer, but now I tested it with a recording made in Unity and it crashed as well, even though it seems that it played back more before crashing.

velenoise, 2018-10-02:
@matkatz Do you have any time estimate on this fix? I'm currently working on a project during part of which I won't have access to the physical devices, so the ability to use the recordings would be essential in my workflow. Is there anything we can do to help on this as well?

velenoise, 2018-10-03:
@RealSense-Customer-Engineering just an FYI: reinstalled the SDK v2.14.1 and playback is working properly now, so I can go with that until we have the correction for the new versions

frenchtouchXR, 2018-10-03:
@velenoise  You're right. The problem is I would really need to use the filters. They're integrated since the 2.15 version. :/

frenchtouchXR, 2018-10-21:
I Simply would like To tell I opened this thread 2 month ago. Do you have the release date of the patch ? Or is there another problem ?

NeuroRanger, 2018-10-28:
I had the same problem. In the latest release the issue seems to be fixed.
I downloaded version 2.16.2.0, compiled it, started a scene in Unity, loaded a recorded .bag-file, it showed the content and didn't crash anymore.
Thanks a lot for fixing that!

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi @frenchtouchXR,

How does v2.16.2 work for you?

frenchtouchXR, 2018-10-30:
Hi,

The record and playback seems to work but the RSdeviceinspector.cs script doesn't seems to work. 

The real interest since version 2.15 is playing with filters. If we can't use them. We came back to 2.14 version.

frenchtouchXR, 2018-10-30:
I opened the RSdeviceinspector.cs script. 

Inside there's a fonction : private IEnumerator WaitForDevice()

And there is this line : 

//rsdevice.OnStart += onStartStreaming;

I just reactivated this lign. Now it works.

frenchtouchXR, 2018-10-30:
OK ! Multicam rec and playback now integrated. 

The unity session is still fragile and crashes from time to time but not systematically when playback. It crashes when I stop the game from time to time. 

I could use the decimation filter. 

Now let's play. Please work on optimising the pointcloud in Unity and stabilising the session. 

Case closed

Thanks

FT-XR

closed 2018-10-30 by frenchtouchXR

metanamo, 2018-11-07:
Hi,
I'm sry to jump up on this issue thread.
Using latest SDK/unitypackage and latest Unity I can't get multi camera streaming to work.
I can't even change the static Instance to a none static one. The _CustomProcessingBlock_ at the end of RsDevice.cs wants static member. 
@frenchtouchXR  Could you please help me? Which version of SDK Unity and Firmware you are using?

Thanks

frenchtouchXR, 2018-11-07:
In the 2.16.3 version, you just need to duplicate the device and the pipeline. Then create a new material for the mapping and link it to the copied elements.

matkatz, 2018-12-04:
Sorry for the long delay.
The Playback issue was handled by PRs #2646, #2598 which were merged to the master branch and are available on release [v2.17.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.17.0)