realsenseai/librealsense #2308: poll_for_frames does not get frames from bag file

chunhai, 2018-08-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.14.5                                     | 
| Platform                                 | odroid  |
| SDK Version                            |  sdk2.0                        | 
| Language                            |  c++                         | 
| Segment			|                       | 

### Issue Description
I try to use rs::config with rs2::pipeline to record and read bag file according to the example codes here : https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md. Record process works well and it can generate bag file which can be loaded by realsense-reviewer. However, when I try to read bag file with pipeline,  pipe.poll_for_frames(&frames) always return false.

Does anyone have a clue about this problem? Thanks.

armandok, 2018-08-29:
I have the same problem on Ubuntu 18.04

dorodnic, 2018-09-04:
There has been several complaints about playback from bag files recently, and we are currently in the middle of a major refactoring to address most of the issues + add some of the requested functionality. I suspect this falls into the same category, and hope to resolve in the next release.

dorodnic, 2018-09-25:
Could you please try again with [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)?

chunhai, 2018-09-26:
My problem was solved, and it was not actually caused by library bugs. I think the tutorial I referenced[https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md](Record and Playback) is kind of misleading.

As claimed there, the following code should replay the bag file, but it cannot because poll_for_frames is non-blocking.

`rs2::config cfg;
cfg.enable_device_from_file("path_to_input_file.bag");
rs2::pipeline pipe;
pipe.start(cfg); //File will be opened in read mode at this point
for (int i = 0; i < 30; i++)
{
    rs2::frameset frames;
    if (pipe.poll_for_frames(&frames))
    {
        //use frames here
    }
    //Note that we use poll_for_frames instead of wait_for_frames
    //This is because the file only contains a finite amount of frames
    // and if we use wait_for_frames then after reading the entire file
    // we will fail to wait for the next frame (and get an exception)
}
pipe.stop(); //File will be closed at this point`

Instead, it can be solved like this
`while (true)
{
   rs2::frameset frames;
    if (!pipe.poll_for_frames(&frames))
    {
        continue;
    }
    //use frames here
}`

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
Hi chunhai,

Glad to hear that your problem was solved.
I am going to close this one for now. Please report back any issues or ask any questions.

Thanks!

closed 2018-10-12 by RealSense-Customer-Engineering