realsenseai/librealsense #2310: Colorspace mix up Realsense Viewer

kevindehecker, 2018-08-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400  | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Linux (Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.13.0-43-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.15                         | 
| Language                            |  N.A. | 
| Segment			|  Realsense viewer                          | 

### Issue Description
When capturing a stream from the color camera on the D435 to a bag file, and selecting the format RS2_FORMAT_BGR8 for the stream:
`cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 15);`

The realsense viewer messes up the color order, when opening that bag file:

[image]

FYI, the small car should be orange instead of blue, the van should be white. The grass should be much greener.

dorodnic, 2018-09-03:
Hi @kevindehecker 
If I understand correctly, this is just the way our Viewer displays BGR. 
When you record BGR, the Viewer will only play BGR (if you open the RGB Camera panel, you should see BGR as the format)
As far as I know, OpenGL don't support native BGR rendering, so converting the texture to RGB prior to rendering would introduce significant CPU overhead. A better solution would probably to use "modern" OpenGL and fix it in a fragment shader, but this option is not available to us right now (would require too much changes in the Viewer)

AntoineWefit, 2018-09-03:
I had a similar issue a while ago : the pictures taken by the camera were blueish...
Maybe you already solved your problem, maybe it is also completely different from mine, but I corrected the weird color simply by replacing : 
`cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 15);`
with
`cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 15);`

kevindehecker, 2018-09-03:
@dorodnic Maybe we can display a warning in the RS viewer that the input stream is detected as none-RGB, but displayed as RGB nonetheless. (it took me a while to figure out the problem was not in my code)
BTW, @AntoineWefit  I actually need BGR because OpenCV uses that as standard (I know, annoying...).

closed 2018-12-14 by RealSense-Customer-Engineering

wutter, 2020-08-11:
Issue still presisstant, any news on this?