realsenseai/librealsense #2314: Make fail

abhileshborode, 2018-08-28:
| Camera Model                       | { D435} | 
| Firmware Version                   | (Open RealSense Viewer --> 2.16.0) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  1.1.2.realsense-1.3.0                                        | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.15.0}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
On following the build instructions  catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release

throws a make error 

/home/abhilesh/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:382:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for schedule(dynamic)
 ^
In file included from /usr/include/librealsense2/hpp/rs_device.hpp:8:0,
                 from /usr/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/include/librealsense2/rs.hpp:9,
                 from /home/abhilesh/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/realsense_node_factory.h:11,
                 from /home/abhilesh/catkin_ws/src/realsense/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/abhilesh/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1:
/usr/include/librealsense2/hpp/rs_sensor.hpp: In instantiation of ‘void rs2::frame_callback<T>::on_frame(rs2_frame*) [with T = rs2::asynchronous_syncer; rs2_frame = rs2_frame]’:
/home/abhilesh/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:1396:1:   required from here
/usr/include/librealsense2/hpp/rs_sensor.hpp:109:13: error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’
             on_frame_function(frame{ fref });
             ^
realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:1188: recipe for target 'realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2

Thanks

whyscience, 2018-08-30:
meet the same error too

neomanic, 2018-08-30:
Also running in to this. Reverting to the pkgs for previous version (`sudo apt install librealsense2-dev=2.15.0-0~realsense0.83` instead of `2.16...`) fixes the compile error.

whyscience, 2018-08-30:
```
ver="2.15.0-0~realsense0.83"
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils=${ver}
sudo apt-get install librealsense2-dev=${ver}
sudo apt-get install librealsense2-dbg=${ver}
```

do the same

dorodnic, 2018-08-30:
ROS repo was updated to resolve this.

kruftindustries, 2018-09-03:
@abhileshborode What did you do to make this work?

yeyanlei, 2018-09-05:
I meet the same problem, how can i solve this problem. Thank you

yeyanlei, 2018-09-05:
hello sir, i meet the same problem(error: no match for call to ‘(rs2::asynchronous_syncer) (rs2::frame)’. I try to solve this problem(ver="2.15.0-0~realsense0.83", sudo apt-get install librealsense2-utils=${ver}), The system display (Depends: librealsense2 (= 2.15.0-0~realsense0.83) but 2.16.0-0~realsense0.85 is to be installed).
Thanks for your help

abhileshborode, 2018-09-06:
@yeyanlei 
Delete the previous clone.
Re-clone and build again.

Hope it helps

yeyanlei, 2018-09-07:
@abhileshborode Thanks for your help, re-clone did not work. ( Build from sources by downloading the latest Intel® RealSense™ SDK 2.0 and follow the instructions under Linux Installation, https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). i don't know why  DCMAKE_BUILD_TYPE=Release: command not found. 
[image]
and realsense-viewer command not found

frederico-klein, 2018-09-07:
i tried using the release 2.0.3 and it failed. I suppose the latest git version has this fix, but it hasn't made it into a static release yet. 
downgrading worked for me with the exception I also needed to downgrade librealsense2 like:
ver="2.15.0-0~realsense0.83"
sudo apt-get install librealsense2=${ver}
sudo apt-get install librealsense2-utils=${ver}
sudo apt-get install librealsense2-dev=${ver}
sudo apt-get install librealsense2-dbg=${ver}

frederico-klein, 2018-09-07:
@yeyanlei maybe you confused a / and \ 
you are separating what is supposed to be a single line as in: 
cmake ../ -DCMAKE_BUILD_TYPE=Release

yeyanlei, 2018-09-09:
@frederico-klein , i don't know why i input realsense-viewer, (realsense-viewer: command not found). i didn't install from Debian Package
[image]
[image]

Omegastick, 2018-09-10:
That sounds like you didn't do `make install`.

yeyanlei, 2018-09-10:
@Omegastick , Thanks.  Could you show me the commands, or where are commands. and tutorials (build from source) didn't mention the realsense-viewer command (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). I just saw it at  (  Install from Debian Package)

RealSense-Customer-Engineering, 2018-09-18:
[Realsense Customer Engineering Team Comment]
Hi @yeyanlei,

Is the issue resolved?
As @Omegastick mentioned, you need to run "$ sudo make install" if you build librealsense from source, then the apps/libraries will be installed in /usr/local. Once the installation done, you can run "$ realsense-viewer" or other apps directly.

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @yeyanlei,

frederico-klein, 2018-09-27:
> [Realsense Customer Engineering Team Comment]
> Hi @yeyanlei,
>
> Is the issue resolved?
> As @Omegastick mentioned, you need to run "$ sudo make install" if you build librealsense from source, then the [...]

This is probably your answer. 

HOWEVER, you might want to use checkinstall instead to keep your system clean like:
`sudo apt-get install checkinstall`
`sudo checkinstall -D make install`

yeyanlei, 2018-11-05:
Thanks @frederico-klein . When I used new SDK version.  Problems were solved