realsenseai/librealsense #2317: About world coordinates of D435 in realsense-viewer

kubrakaracan, 2018-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.09.14.00) | 
| Operating System & Version |    Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.15.32)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<15> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, in projection in Realsense SDK 2.0 it is written that the positive x-axis points to the right, the positive y-axis points down, and the positive z-axis points forward. However, in realsense-viewer the coordinate system looks different ( x to left, y points up). What is the reason for that? I also shared a ss for making my question clear. Thanks in advance. 
[image]

AntoineWefit, 2018-08-28:
I don't have the answer for this question, but the I noticed that the axis orientation is also not the same between the Viewer visualization and the point cloud capture itself. So, as you said, there is indeed a difference with the different axis definitions.

Point cloud in the Viewer (lets assume X goes to the right, Y up and Z forward for the depth) : 
[image]

Same point cloud in Meshlab (the coordinates are not really what you would expect from the Viewer as Y is pointing down, but they match with what the documentation says) : 
[image]

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
The points are calculated by rs2_deproject_pixel_to_point.
https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L50

and in other hands, the green axis points to Y-, in the realsense-viewer.
https://github.com/IntelRealSense/librealsense/blob/master/common/rendering.h#L1210

That is the difference what you see.

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@kubrakaracan  Any other questions about this ticket? Looking forward to your update. Thanks!

kubrakaracan, 2018-11-04:
Thank you for your answer! but I meant to say that why y-axis points to upward in realsense-viewer render? Should it point to downward as written in the document? I also checked the rendering algorithm you posted, and it is also written in positive manner. if y-axis points upward then x-axis in realsense-viewer points inversely?

kubrakaracan, 2018-11-05:
I got the point now, thank you so much !!

closed 2018-11-05 by kubrakaracan