realsenseai/librealsense #2319: Multiple Cameras for Extended Duration

upxela, 2018-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (5.10.3.0) | 
| Operating System & Version |   {Windows 10 }  | 
| Platform                                 | PC |
| SDK Version                            |  {2.14}                          | 
| Language                            |  {python2.7 }                          | 

### Issue Description
Does anybody have experience recording from multiple (e.g. 3+) D400 cameras for longer duration? I realize that I start dropping frames recording depth stream at 1280x720 resolution, 30fps after 40 seconds of recording when recording with 4 D435 cameras, and after approximately 120 seconds with 3 D435 cameras at the same resolution and framerate.

MartyG-RealSense, 2018-08-28:
There was a discussion about this on Intel Support a couple of weeks ago.

https://communities.intel.com/message/560215#560215

The short answer is that you should be able to increase your chances of stability when running multiple cameras at the same time for a prolonged period if you use an industrial-grade USB hub.  Industrial-grade USB cables may not hurt either.  

Industrial grade USB equipment is used in critical applications that cannot afford to have dropped frames, such as medical scanners.  A company that can supply industrial-grade equipment compatible with the 400 Series cameras is Newnex.

upxela, 2018-08-29:
@MartyG-RealSense 

I realized the USB hubs I was using were the cause of this behavior. Thanks for your advice!

closed 2018-08-29 by upxela