realsenseai/librealsense #2321: Low fps when aligning depth to colored frame

armandok, 2018-08-29:
Required info |  *
------------ | -------------
OS | Ubuntu 18.04
Platform | C++
Resolution | 640*480 @ 30fps
SDK version | 2.10.16 (source)

I've been trying to record the aligned color and depth frames on disk using different ways but the problem is that I get around 10 fps of images since I think alignment takes more than 33 ms.
Here is my code:
```cpp
rs2::pipeline pipe;
rs2::config conf;
conf.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
conf.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
rs2::pipeline_profile profile = pipe.start(conf);

// Initialize a vector that holds filters and their options
rs2::frame_queue filtered_data;
rs2::align align(RS2_STREAM_COLOR);
bool stopped = false;

std::thread processing_thread([&]() {
    rs2::processing_block frame_processor(
            [&](rs2::frameset data, // Input frameset (from the pipeline)
                rs2::frame_source& source) // Frame pool that can allocate new frames
            {
                rs2::frameset processed = align.process(data);
                rs2::video_frame other_frame = processed.first(align_to);
                rs2::depth_frame aligned_depth_frame = processed.get_depth_frame();

                if (!aligned_depth_frame || !other_frame)
                {
                    return;
                }

                rs2::frame filtered = aligned_depth_frame; // Does not copy the frame, only adds a reference

                rs2::frameset combined = source.allocate_composite_frame({ filtered, other_frame });
                source.frame_ready(combined);
            });

    frame_processor >> filtered_data;

    while (!stopped) //While application is running
    {
        rs2::frameset fs;
            if (fs = pipe.wait_for_frames()) frame_processor.invoke(fs);
    }

});

while (app) {
rs2::frameset frames = filtered_data.wait_for_frame();
rs2::video_frame post_processed_image_frame = frames.get_color_frame();
rs2::depth_frame post_processed_depth_frame = frames.get_depth_frame();

    stbi_write_png("SomeImageName.png", post_processed_image_frame.get_width(), post_processed_image_frame.get_height(),
                   post_processed_image_frame.get_bytes_per_pixel(), post_processed_image_frame.get_data(), post_processed_image_frame.get_stride_in_bytes());

    // Creating OpenCV matrix from depth image
    Mat depth(Size(640, 480), CV_16UC1, (void*)post_processed_depth_frame.get_data(), Mat::AUTO_STEP);
    cv::imwrite("SomeImageNameD.png",depth);

}
stopped = true;
processing_thread.join();
return 0;
```
I've tried this without the processing block and without an additional thread but they are essentially the same.
I think I need to wait for frames in the main thread and then enqueue the frames and do the alignment in another (possibly multiple other) thread(s).
If this is the right way to go, how can I enqueue framesets? If not, what should I do?
P.S. does anyone know a better way to save the depth frame on disk? (I don't want to colorize it)

armandok, 2018-08-29:
Hi @dorodnic
Thanks for the fast reply.
Yes, the `align` example is also slow. I thought this is how it should be!
I am running a 64-bit **Ubuntu 18.04.1 LTS** on my laptop and these are my specs:

Hardware | Value
--- | --- 
CPU |  Intel® Core™ i7-8700 CPU @ 3.20GHz × 12
Memory | 32 GiB

I recompiled the library with `-DCMAKE_BUILD_TYPE=Release` option but I see no change. I did also try the precompiled one from `apt-get` and that is also the same.
The `realsense-viewer` runs fine and fast, so I really doubt this is due to the auto-exposure.

armandok, 2018-08-29:
I also tried to save the frames in a rosbag file, so that I can load them later and do the alignment.
But the problem was that it couldn't read the frames from the rosbag file. I think I've tried everything yet I cannot record depth and color frames at 30 FPS.

dorodnic, 2018-08-29:
This is very strange, compared to the last measurements we did of `align` ([#2040](https://github.com/IntelRealSense/librealsense/pull/2040))

armandok, 2018-08-30:
I tried the `align` example on a computer with **Ubuntu 16.04** and it ran as expected.
Maybe there is a problem with 18.04?

dorodnic, 2018-08-30:
We did a considerable amount of testing with 18.04 recently, @ev-mp could you comment if you encountered something similar?

mfassler, 2018-09-05:
I'm also seeing slowness / glitchiness.  It used to be consistently slow fps, but since I've pulled the latest version from Github, now it's better, but still a little bit "glitchy".

I've got some Python code.  I time this function:
```
t0 = time.time()
aligned_frames = align.process(frames)
t1 = time.time()
print(t1-t0)
```

If I playback from a rosbag file (created using the librealsense config.enable_record_to_file function, playback using the config.enable_device_from_file function) then the time is typically about 0.005 seconds.  But if I run from a real device, then sometimes the time is up to 0.12 seconds, which creates a noticeable "glitch" in the playback.

(Running rs-align built with "-D CMAKE_BUILD_TYPE=RELEASE" also seems glitchy to me.)

I have attached two images:  the histograms of the time to do align.process() in Python.  One is playback from a rosbag, one is live from real device.  

#### Histogram from rosbag (times to process 591 frames): ####
[image]

#### Histogram from live device (times to process 522 frames): ####
[image]

You can see that, sometimes, it takes about 0.12 seconds to process a frame, which causes a noticeable glitch.  

(This is Ubuntu 16.04, on a recent i5-8400 cpu, 32 GB ram, GeForce GTX 1080 video)

(again, I used to consistently get slow fps, so this is an improvement...)

mfassler, 2018-09-05:
@armandok I noticed that you're using librealsense version 2.10.  There were some perfomance improvements around v2.14, in July.

armandok, 2018-09-05:
@mfassler Thanks for the suggestion. I'll try it and let you know.

WeiChihChern, 2018-09-07:
@dorodnic I've same low fps issue when using align function.  I follow the align example from the SDK documentation. 

Windows 10, VS 15, RealSense 2.15.0, C++, D415
Spec: i5-8400, gtx 1080
```
s2_stream align_to = find_stream_to_align(profile.get_streams());  // Find a stream for alignment, color stream is preferred
rs2::align align(align_to);

while ... /* In side a while loop capturing frame */
data = align.process(data);                      	// Align (rectified) depth and color frame
rs2::video_frame color = data.first_or_default(align_to);
rs2::depth_frame depth = colorizer(data.get_depth_frame());
```
Low fps problem happens under both Debug and Release mode.  When not aligning (only capturing frames), it becomes smooth again.

dorodnic, 2018-09-12:
Just trying to make sense of this thread:
* @armandok was reporting low-fps on Ubuntu 18 that was resolved after moving to 16. 
* @WeiChihChern - you are seeing both low fps and high CPU utilization (#2376) - do you have `-DBUILD_WITH_OPENMP=true` in Cmake options?
* @mfassler is reporting occasional slowness (glitches) with live device (but not recording). 

Is this accurate?

mfassler, 2018-09-13:
@dorodnic Yes.   (also:  in an older version, I used to get consistently slow FPS for align.process(), but in the latest github version it's only the occasional glitches.)

dorodnic, 2018-09-13:
Thanks, we will continue to look into each of the issues.

shinstra, 2018-09-26:
I am also encountering this issue and struggling to get 4fps. My script is a cut down version of the align-depth2colour example (after the aligned frames are retrieved I continue to the next iteration of the loop without the additional processing). I have tried compiling both with and without openmp.

Sys info:
* Win 10
* CPU: i7-7700K
* GPU: 1080Ti
* librealsense: v2.16.0
* firmware: 05.10.03.00
* Python

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@armandok  Do you still have Low fps issue with the latest librealsense v2.16.1 on Ubuntu 18?

armandok, 2018-10-19:
@RealSense-Customer-Engineering Yes, I do still have this issue on 2.16.1. I'm getting around 14 fps

ev-mp, 2018-11-19:
Hello @armandok ,
The latest version added CUDA implementation for rs2::align functionality.
Please check and update

mfassler, 2018-11-30:
The latest build from Github (v2.17.0) is no longer glitchy for me.  rs.align() is nice and smooth.

armandok, 2018-12-04:
Hi @ev-mp 
I tried 2.17 and the rs-align is much much faster right now and is smooth.
When I try my own code (included at the top of this page), the framerate is now 20 fps.
It is still not 30 fps but it is twice what it was before.

RealSense-Customer-Engineering, 2018-12-04:
[Realsense Customer Engineering Team Comment]
Hi @armandok,

I have run your code on my laptop.
the `processing_thread` thread is working in 30fps. but the function `stbi_write_png` and `cv::imwrite` kill too much time.

xiongdahua555, 2018-12-05:
I think threr is no need to align each time, internal and external parameter is fixed.

armandok, 2018-12-25:
@RealSense-Customer-Engineering I replaced the `stbi_write_png` with `cv::imwrite` and now I'm getting 25 fps! Thank you.
I guess this not an issue anymore.

closed 2018-12-25 by armandok

zofnatb, 2019-11-05:
I have the same problem. Low fps when aligning depth to colored frame. 
in resolution : 1280X720 and (USB 3.2)  - 6fps
At lower resolution it 18 fps
I use realsense D435.
Librealsense v2.17 on Ubuntu 16.4

work in c++ with qt. 

Thank you