realsenseai/librealsense #2324: Colorizer use since 2.16

edmBernard, 2018-08-29:
I use a D435 realsense camera.

on sdk 2.15, we can apply colormap on depth camera this way: 
```cpp
rs2::colorizer color_map;
rs2::frame depth;
depth = colormap(data.get_depth_frame());
```
It don't seem to work any more on 2.16 I got this error : 
```
/home/user/Documents/test_realsense/src/rs-capture.cpp:113:49: error: no match for call to ‘(rs2::colorizer) (rs2::depth_frame)’
         depth = color_map(data.get_depth_frame());
```

This methode from sdk test case still work:
```cpp
rs2::colorizer color_map;
rs2::frame depth;
depth = data.get_depth_frame().apply_filter(color_map);
```

This change was not mentioned in changelog so it's maybe a bug : https://communities.intel.com/thread/128477

dorodnic, 2018-08-29:
Hi @edmBernard 
You are right, we should add it to the C++ API changes doc.
You can still use the `colorize` function like in all the samples, and the new `apply_filter` function. 
`operator()` was just a convenience function at the C++ wrapper level, and I personally find the explicit call to `colorize` more readable. As for `apply_filter`, it creates a unified way you can apply any processing block. 
Would you prefer the `operator()` back in the API?

edmBernard, 2018-08-29:
It's ok for the change if it's noted in API change. 
When I post my issue, I didn't see you have update examples too. So I think it's fine.

closed 2018-08-29 by edmBernard

dorodnic, 2018-08-29:
I've updated the [API Changes](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#from-version-2150-to-2160) page accordingly