realsenseai/librealsense #2326: Unindexed Bag

upxela, 2018-08-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.3.0 | 
| Operating System & Version |  Windows 10 | 
| Platform                                 | PC  |
| SDK Version                            | 2.14                          | 
| Language                            |  python2.7                    | 

### Issue Description
After recording .bag files, I have had trouble opening them up. Specifically, I get the following error message: 

> RuntimeError: Failed to resolve request. Request to enable_device_from_file("file.bag") was invalid, Reason: Failed to create ros reader: Bag unindexed

I read on #1857 that this error usually means that there is a problem with closing the recording. I don't however see a dispose function in pyrealsense2; is there something else I should be aware of to close the recording properly?

dorodnic, 2018-08-30:
Hi @upxela 
If you are using `pipeline` I think that `stop` should be closing the `bag` file. With `sensor` it is probably `close` method.

upxela, 2018-08-30:
@dorodnic 

I realized I had made the config variable a property of a class. By making it a local variable instead, everything closes out properly and I can open the bag file without any problems.

Thanks!

closed 2018-08-30 by upxela