realsenseai/librealsense #2329: How can I calibrate D415 or D435 ?

blackball, 2018-08-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | Latest | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    | 4.9.8                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

How does D400 cameras calibrated in factory ? Are they all well taken cared ? I have bought both D415 and D435, but the default calibration seems not as accurate as I want. I decided to calibrate the cameras by myself. 

I have read the custom calibration white paper (https://www.intel.com/content/www/us/en/support/articles/000026725/emerging-technologies/intel-realsense-technology.html), It's obviously outdated since I could not find the tool and pattern image it mentioned. 

I also downloaded and installed the calibration tools (https://downloadcenter.intel.com/download/27955/Intel-RealSense-D400-Series-Calibration-Tools-and-API), inside it, there's a manual to tell the process. It uses a pretty small calibration pattern that user could use a iphone to display it and use as the calibration board. I really doubt if it could provide good accuracy for the calibration. Or I misunderstood something. 

If anybody could point me a right direction to do the custom calibration, It will be hugely appreciated. Thanks!

MartyG-RealSense, 2018-08-30:
RealSense cameras are usually well calibrated when purchased.  Circumstances in which the camera may need calibrating after purchase is if the camera experiences a fall to the ground or hard knock.

If you want to use the default calibration tool, and do not need to write your own custom calibration app, you should download the file 'CalibrationTool-2.6.4.0-Setup.exe' on the tools download page that you linked to.  It is the lower of the two links in the side-panel.

The calibration checker board image was changed around June 2018.  You should now use the one that is linked to from the downloads page.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/print-target-fixed-width.pdf

closed 2018-09-09 by blackball

octalframes, 2019-11-27:
OEM Calibration tool library/code should be made public. Intel does not have service centers. For these type of problems we need ship it back to distributors. Even after buying product if we cannot calibrate the intrinsic values then there is not point of continuing with this product. Intel cannot except us to spend another $1500 to buy it.

MartyG-RealSense, 2019-11-27:
In 99% of cases it is not likely to be necessary to calibrate the intrinsics, as it is the extrinsics that have the most effect on the output image quality.  

If calibration does not correct a problem with output image then using the 'gold reset' function of the dynamic calibration API's CustomRW program to reset the camera to factory defaults may help.  

I have previously submitted a suggestion of making the OEM calibrator the standard free version (the source code would not be released) but the decision is not up to me.

d3-worgan, 2020-07-17:
Hi thanks everybody,

Can I just confirm @MartyG-RealSense that the "gold reset" will restore the device to the factory calibration settings? Can the "gold reset" be done in the intel realsense viewer?

Can I also confirm that if the camera has taken a bump then the factory settings would not work anymore? (potentially)

Thanks!

MartyG-RealSense, 2020-07-17:
Hi @d3-worgan   Yes, a "gold reset" should reset the camera to its factory settings.  It cannot currently be done in the RealSense Viewer though.

Instead, calibration software called the **Dynamic Calibrator** should be installed.   It contains a program called **CustomRW**.  Launching this program with the command below initiates a Gold Reset of the camera:

 **Intel.Realsense.CustomRW - g**

The Windows version of the Dynamic Calibrator tool can be downloaded from here:

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

Instructions for installing the Linux version of the tool are on page 14 onward of the user guide for the tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

A bump will not affect the factory calibration data stored in the camera.

PierreLouisT, 2021-12-28:
Hello everybody ! I am a newby here and thanks for give me the opportunity to ask some questions. 
I am using a Realsense D435 and I need in priority to calibrate my camera. I already used the Dynamic calibrator tool and the on-chip calibrator (respectively to get extrinsics and intrinsics) with the document at this link : https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

 I also have done some calibrations using tutorials for openCV : https://docs.opencv.org/3.4/dc/dbb/tutorial_py_calibration.html
However I need to use the CustomCalibration available in this document : https://dev.intelrealsense.com/docs/d400-series-custom-calibration-white-paper.
As far as know it is not necessery to use it if I already installed Dynamic calibrator and the on-chip calibrator but I really need it. 

I have the Custom Calibration folder inside calibrationtoolAPI  according to 7.2 part API Package structure in the firt link. I compiled the example provided for the dynamicCalibrator but I dont know how to compile it and use it to have the same setup for the CustomCalibration here : 
https://www.intel.com/content/www/us/en/support/articles/000026725/emerging-technologies/intel-realsense-technology.html
Another proble is that when I compile, there is not this library : #include "DSDynamicCalibration.h"

I apologize if I make english mistakes, I am from France 

Thanks in advance for all the help you could bring me :)

MartyG-RealSense, 2021-12-28:
Hi @PierreLouisT   Are you using the Windows Custom Calibration API compressed file **DynamicCalibrationAPI.zip** and have extracted the folder inside the zip file?

If you are and you have been able to extract the **DynamicCalibrationAPI** folder, is the file 'DSDynamicCalibration.h' located in the **Includes** sub-directory?

[image]

In regard to your other question, could you provide more information please about the 'same setup' that you want to achieve with your compiled custom API example?

PierreLouisT, 2021-12-28:
> Hi @PierreLouisT Are you using the Windows Custom Calibration API compressed file **DynamicCalibrationAPI.zip** and have extracted the folder inside the zip file?
>
> If you are and you have been able [...]

I want to calibrate my camera using custom calibration and have that : 
[image]

but I am using linux and not windows so I think it is a good idea to compile the customcalibrator inside the DynamicCalibration tool but I got this error : 
[image]
I got some library problems as missing library, I saved all of them inside the common folder in examples and it worked. 
The only thing I need to know is how to avoid that error. 
I checked and all of your documents.h are available inside the dynamicCalibrationAPI. 

I hope it is more clear for you, thanks a lot for your help :-)

MartyG-RealSense, 2021-12-28:
Page 19 of the custom calibration white paper states that **OpenCV 3.3.1** needs to be installed in order to use the calibration example.  Have you got this OpenCV version installed, please?

[image]

huonggiangdhtn, 2022-04-06:
[image]
I use 2 D435 camera. They have the same y and z value. They are separate 25 cm, no rotation, like in the second image. But when I build poitcloud it is not good. I have check again the camera parameters, they are at the same of manufacturer values.
[image]
I don't know where to ask, so i post here.

MartyG-RealSense, 2022-04-06:
Hi @huonggiangdhtn   When you have multiple 400 Series cameras generating a pointcloud, their clouds are not automatically stitched together and so if you are aiming for a single large cloud then you have to combine the clouds together.

If you are not stitching the clouds together already, one way to do so is to perform an **affine transform** to set all of the individual point clouds to the same position and rotation in 3D space.  A RealSense SDK instruction that can do this is **rs2_transform_point_to_point()**.  It uses a 3x3 rotation matrix and a 3-component translation vector.

Alternatively, Intel provide a guide for stitching multiple RealSense point clouds together in **ROS**

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

MartyG-RealSense, 2022-04-08:
As you are using PCL and need to stitch the two pointclouds together, an appropriate way to do that may be to use PCL's pointcloud **registration** via Iterative Closest Point  (ICP), as described in the PCL tutorial in the link below.

https://pcl.readthedocs.io/projects/tutorials/en/latest/pairwise_incremental_registration.html

huonggiangdhtn, 2022-04-12:
Thank you very much, it is very useful for me. And here is the best result I have. There is still error in this pointcloud. I want to ask this is normally when we combine different camera, or I still do not good my job. Because i don't have much experience on this.
[image]

MartyG-RealSense, 2022-04-12:
Don't worry, you are doing well.  :)

The stitched image above is much better now.  There is a clear distortion on the point cloud of one of the cameras though that is making the chessboard look as though the paper is squashed up.  

[image]

In reality, it may look like that because there is an error that is making part of the chessboard be overlaid on top of the image and the image would likely look correct if that overlay was not present.  The overlaid part resembles the piece from your previous stitch attempt.

[image]

This makes me think that there may be a problem in the script and not a mis-calibration of that particular camera.

afenghandsome, 2022-04-13:
hello ,I'm new  learner. I can not find CustomCalibrationTest.exe in my download file.

afenghandsome, 2022-04-13:
[image]

afenghandsome, 2022-04-13:
@MartyG-RealSense

huonggiangdhtn, 2022-04-13:
Can I ask some more question? Can I use se PCL's pointcloud registration as a way to find transform matrix? Is it a good way? With the best result when I apply with tree segmentation, I still get error. You see, there is one purple stem but it become two. My rgb image go through a model to get segmentation image . In my app, there is a code to transform from depth image to pointcloud, I think it is good because this code is popular. Can  you give me some comments of what way to deal with it, or another way to get better tranform matrix, or I  should change the type of camera. I'm very stressed on this for weeks.
[image]

MartyG-RealSense, 2022-04-13:
Hi @afenghandsome  Page 44 of Intel's **Custom Calibration White Paper** document states that you build the DynamicCalibration example's .exe yourself by opening **CustomCalibrationTest.sln** under examples\CustomCalibration with Visual Studio and build “Release”  “x64”.

https://dev.intelrealsense.com/docs/d400-series-custom-calibration-white-paper

afenghandsome, 2022-04-13:
yes ,i try it .i also configure opencv3.3.1, but the CustomCalibrationTest can not work well.I search some methods,but i can not solve.
[image]
[image]
@MartyG-RealSense

MartyG-RealSense, 2022-04-13:
@huonggiangdhtn  Would the PCL registration function **TransformationEstimation** meet your needs for finding the transform matrix, please?

https://pointclouds.org/documentation/classpcl_1_1registration_1_1_transformation_estimation.html

MartyG-RealSense, 2022-04-13:
@afenghandsome  Apparently the **LNK118** error in your Visual Studio console refers to **cannot open input file**, suggesting that it is having a problem with opening a file called 'libpng.lib'.

When the RealSense SDK first launched at the end of 2017, it was using OpenCV 3.4 for its examples.  So it may be worth trying upgrading your OpenCV from 3.3.1 to 3.4 if possible.

afenghandsome, 2022-04-13:
I am try opencv3.4 ,but it still keep on “could not open file ”libpng.lib" ",I also try opencv4.5, but i can not solve.
@MartyG-RealSense

afenghandsome, 2022-04-15:
Yes, if you want to get the EXE file, you must compile it, but I have tried many methods and still can't solve the error problem.
@MartyG-RealSense

MartyG-RealSense, 2022-04-15:
According to the Custom Calibration white-paper, an additional step for building CustomCalibrationText.exe is to place the OpenCV headers and libraries under the folder location **CalibrationToolAPI > [version number]**.  The notes are for the 2.8.0.0 version though, stating that the files should be under **CalibrationToolAPI > 2.8.0.0**.

[image]

The equivalent location in the current version of the tool would be **CalibrationToolAPI > 2.11.0.0**.  As the notes were designed for 2.8.0.0 though, I can offer no guarantee that placing the OpenCV headers and libraries under the 2.11.0.0 will enable the example to build.

[image]

InTo-7, 2025-05-19:
Hi,

@MartyG-RealSense 

I am using an Intel Realsense D457 for research.

My goal is to smooth the edges of spheres and other objects in the depth image.

If there is no physical fault in the camera, does that mean there is no point in doing calibration?

If you are going to calibrate, do on-chip calibration, focal length calibration, tare calibration, dynamic calibration. I would like to know what the difference is between these and the order in which they should be performed.

Calibrate the D457 using the following as a reference.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

I used Intel Realsense Viewer v2,551 and Intel Realsense Dynamic Calibrator v2,14,2,0.

Could you please investigate this issue?

Thank you very much.

MartyG-RealSense, 2025-05-19:
Hi @InTo-7   Usually a RealSense camera will be well-calibrated already unless it has received a physical shock, such as a hard knock, a drop on the ground or severe vibration.

If you have rough / sharp edges then instead of performing a calibration,  you may be able to rectify that using a **Visual Preset**, which applies a range of camera configuration settings that balance the camera to favor certain effects, such as cleaner edges.  The **Medium Density** preset is one that I would recommend for smoothing curved edges.  

In regard to the calibration methods and differences between them: the On-Chip calibration mode only calibrates depth, whilst Dynamic Calibration calibrates both depth and RGB.  

On-Chip improves the quality of a depth image, whilst Tare improves its depth measurement accuracy.  

In regard to focal length calibration, ideally the focal length of the camera's left and right sensors will be identical, but when they are not identical and negative effects are caused as a result, left and right focal length can be adjusted with the focal length calibration function to correct this imbalance.

InTo-7, 2025-05-19:
@MartyG-RealSense Thank you for your reply.

1. Does performing dynamic calibration calibrate lens distortion? I tried it, but I didn't understand what parameters were changing and what was being calibrated.

2. If I were to perform all the calibrations, is there an order in which I should do them?

If there are any documents that provide details on the above calibration, please let me know.

MartyG-RealSense, 2025-05-19:
A user guide for Dynamic Calibration can be found here:

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Whilst the guide for On-Chip, Tare and Focal Length calibration is here:

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras1. 

1.  It can correct lens distortion - for example, if the Depth Module circuit board version of RealSense cameras was flexed whilst held in the hands and the sensors had distortions as a result of that board flexing.

In regard to what parameters are changed, that is a complex topic best described by the user guides above.  But you can get a simple insight into parameters by opening the RealSense Viewer tool, going to the **More** option near the top of the options side-panel and selecting the **Calibration Data** drop-down menu option.

https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 has instructions for how the Calibration Data window can be used to reset a camera's calibration to its default factory-new values.  This can be an easy way to improve calibration instead of performing a calibration procedure.
 
2.  I would recommend choosing either Dynamic Calibration or On-Chip / Tare / Focal Length but not using Dynamic Calibration if you use one of the other three calibration methods.  If you use Dynamic Calibration then only use that exclusively and not the others.