realsenseai/librealsense #2330: get depth for each point in rgb picture

123liluky, 2018-08-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | () | 
| Operating System & Version |   {Win10}| 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            |  { 2.14 }                          | 
| Language                            |  {python }                          | 
| Segment			|                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I used code below to show depth of the red dot, are there any problems in the code? I find the value inormally turns to 0 meter sometimes. Thank you very much?
import pyrealsense2 as rs
import numpy as np
import cv2
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
profile = pipeline.start(config)
align_to = rs.stream.color
align = rs.align(align_to)
try:
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
        if not aligned_depth_frame or not color_frame:
            continue
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        depth = depth_image.astype(float)
        depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
        depth = depth * depth_scale
       
        color_image = np.asanyarray(color_frame.get_data())
        cv2.circle(color_image, (320,240),3,(0,0,255),5)
        cv2.putText(color_image, str(depth[320,240],5)+'m' , (330,230), cv2.FONT_HERSHEY_TRIPLEX, 0.5, (0, 0, 255), 1,False)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        cv2.waitKey(1)
finally:
    pipeline.stop()
[image]

dorodnic, 2018-08-30:
Hi @123liluky 
Looking at the code, everything seems right. Please consider that there is some physical displacement between color and depth sensors, so assuming the object is not flat, there are likely to be points on the boundary that are visible by one sensor but occluded by the other. 
When this is the case, (or in some other less likely corner cases) depth value can turn 0.
One solution we offer in the SDK is the post-processing suite. You can configure hole-filling filter if "nearest valid depth" makes sense in your algorithm. You can try in online in the depth-filter notebook [image]

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
Hi @123liluky ,

Is there a still issue here?

closed 2018-11-06 by RealSense-Customer-Engineering