realsenseai/librealsense #2336: Can't open Labview pointcould example

Fangruiding, 2018-08-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   {Win (10)  64bits  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.11.0 }                          | 
| Language                            |  {labview}                          | 
| Segment			|  {VR}                          | 

### Issue Description
Hi, thanks for providing us with Labview wrapper. I have a trouble when opening the point-cloud-plus example.  As you can see in the following pics:
[image]
[image]

Is it a Labview software issue? Or the issue you can help me to solve?  Thanks.

Jake

MartyG-RealSense, 2018-08-31:
Accounts of this error in LabVIEW go back as far as 2005, so finding a useful answer on the internet is rather hard.  The best resource I found about this error was a help guide on the LabVIEW website.

https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019MJqSAM

Fangruiding, 2018-08-31:
Thank you very much!

MartyG-RealSense, 2018-08-31:
If that doesn't work out (the folder structure quoted in the article doesn't sound like yours, as your folder address doesn't have DAQ in it), the example below in LabVIEW 2017 more closely resembles your error message.  I'm sorry I don't have a more definite solution to offer.

https://forums.ni.com/t5/LabVIEW/Failure-to-initialize-lvanlys-DLL/td-p/3759330

Fangruiding, 2018-08-31:
You are very helpful. I am trying and would like to give the solution later.

MartyG-RealSense, 2018-08-31:
Do you get the error if you use the basic Point Cloud demo instead of Point Cloud Plus, please?

Of the functions circled in red, the most useful ones are the intrinsics and extrinsics.  You can learn about these in the link below.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#intrinsic-camera-parameters

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#intrinsic-camera-parameters

Fangruiding, 2018-08-31:
Hi sir, how can I find the basic point cloud demo...... I find everything in the wrapper without seeing it.

MartyG-RealSense, 2018-08-31:
It is called 'RealSense Hello World: Point Cloud'.  The instructions say it can be found with the other demos by unzipping the package called RealSense SDK2 For LabVIEW.  The download link for the package is near the top of the LabVIEW wrapper page.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/labview

Fangruiding, 2018-08-31:
Sir, I don't know why but I can use that resolution now....... Sometimes it still pops out that warning but in most time, it works fine. Thank.

Another question is: what the use of **texture button** in the point cloud plus interface? When I turn it off, the point cloud become much more dense.
[image]

[image]

I have to see, I like you guys and your demos are awesome......

MartyG-RealSense, 2018-08-31:
Sorry for the delay in responding.  I thought very hard about your question.  But it is better to just say "I don't know" than try to give a reply that might be wrong. :D

Fangruiding, 2018-08-31:
Thanks, I think it's better just to turn it off......

Fangruiding, 2018-09-04:
Thank you very much! But whenever I use the texture button, I can barely see any point cloud on the screen. Like the comparison between:
[image]

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
![texture button on](https://user-images.githubusercontent.com/42870295/45037843-88c9b880-b025-11e8-9f0c-b3ad7d0b298c.jpg)

agrunnet, 2018-09-04:
You are using the D435 which is a monochrome imager. It is quite dark (which is fine for depth) but you can go ahead and increase the gain or exposure a bit to increase brightness or you can simply programmatically increase the monochrome brightness by multiplying array by a factor like 3.

Fangruiding, 2018-09-04:
Thanks！So I know there are many benefits to use left camera instead of the 3rd color camera. But if I just wanna figure out proper spatial and down-sample filter parameters, is there any advantage or necessity to use the left camera over the 3rd colored camera?

Fangruiding, 2018-09-04:
Thanks, you guys are so efficient.

closed 2018-09-04 by Fangruiding