realsenseai/librealsense #2343: histogram equalization

123liluky, 2018-09-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {win10} 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  {2.0}                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, how can I set histogram equalization options while displaying depth_colormap in python? I used sample codes below without histogram equalization.
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)        
        images = np.hstack((color_image, depth_colormap))
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)

dorodnic, 2018-09-03:
Well, the simplest way is to use the `colorizer` class, like in this tutorial: 
[Depth to Object](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb)

123liluky, 2018-09-03:
Thank you for helping me solve the problem. I have another doubt now. I put an image which contains a bar with 10mm width and 100mm height in front of D435 camera. The bar in the image is like below:
[image]
In the above image, we can see three dots with different pixel corrdinates. Now I convert pixel coordinates to point coordinates with codes like below:

depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
depth = depth_frame.get_distance(col, row)
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, [col, row], depth)
text = "%.5lf, %.5lf, %.5lf\n" % (depth_point[0], depth_point[1], depth_point[2]) 
print(text)

the result is:
[image]

The problem is that dots [394,130],[394,260] should share the same x and z, while the difference value of y should equal to 100mm. But we can see from the last two lines of the above picture, the x y z values are not in my imagination. However, the values in the firse two lines are in my imagination. They have same y and z while diffrence value of x is nearly 10mm. Are there any problems in the codes above?

ev-mp, 2018-09-06:
Hello @123liluky ,
Deprojecting 2D depth to 3D point must use the intrinsic data of the depth stream.

Note that there is something strange about the marked points:
For the 3rd point the measured depth is 51.4cm, while for the other two points the depth is ~32cm.
Note that deproject does not modify the Z value, so the ~20cm offset is in the original depth.

 You haven't provided a snapshot of the scene, but if a rectangular object was positioned perpendicular to the camera then the depth value for all corners should have been almost the same. Can you provide some data or to explain how these points were selected?

123liluky, 2018-09-06:
Hi, @ev-mp  I put the printed bar perpendicularly in front of the camera and foun pixel coordinates of three points of the bar in RealSense Viewer. The pixel coordinates are [380,130] [394,130] and [394,260]. Then I used codes above to print 3d point coordinates of the three points.

ev-mp, 2018-09-06:
Hello @123liluky ,
The red circle is appear outside of the camera's body which is not consistent with the spec as the origin of 3D should coincide with the physical location of the depth sensor ,so the following statement is not clear:
>I confirmed [0,0,0] in 3d point coordinates through measurement

It would be easier to assist if you were able to provide a depth scene capture using the viewer's "snapshot" or "record" utilities.

As for the marked part - it is  the RGB complimentary sensor.
The Depth sensor origin is marked below:
[image]
You may find the full spec on the [official site](https://www.intel.com/content/www/us/en/support/articles/000026827/emerging-technologies/intel-realsense-technology.html)

123liluky, 2018-09-06:
@ev-mp But in #1904  I find that [0,0,0] is also floating in the air above the camera.
[image]

closed 2018-11-06 by RealSense-Customer-Engineering