realsenseai/librealsense #2345: Connection loss after long term running. (<=2h)

goetzmi, 2018-09-03:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.10.3.0 | 
| Operating System & Version |  Windows 10 (x64)  | 
| Kernel Version (Linux Only)    | | 
| Platform                                 | PC |
| SDK Version                            | 2.15.0.163  | 
| Language                            | C++/CLI | 
| Segment			| Robot | 

### Issue Description

I`m running the camera in a C++ application (Compiler is VS2015) just for getting the color and
depth images. All process after acquiring images is done on the GPU (Geforce GTX1080 TI).
The device runs well, but after some time (mostly after around 2h) it disconnects, the windows
sound for USB deplugging plays, and the following exception ist thrown:

Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: librealsense::windows_backend_exception bei Speicherort 0x000000F0BBAFCF50.
Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: std::runtime_error bei Speicherort 0x000000F0BBAFEE88.
Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: [rethrow] bei Speicherort 0x0000000000000000.
Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: std::runtime_error bei Speicherort 0x000000F0BBAFF1D0.
Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: [rethrow] bei Speicherort 0x0000000000000000.
Ausnahme ausgelöst bei 0x00007FFB76613C58 in NativeApplicationTest.exe: Microsoft C++-Ausnahme: std::exception bei Speicherort 0x000000F0BBAFF348.

The exception occurs in the first of the following codelines:

rs2_process_frame(align, frames, &e);
if (check_error(e) == -1) {
	rs2_release_frame(frames);
	throw exception("Error HelperRealsense - rs2_process_frame");
}

The hardware reset implementation doesn`t work after this exception too.

dorodnic, 2018-09-04:
The main question to focus on is what causes the hardware disconnections. Internally, the hardware / software is being tested for long periods (days) for stability, and while not every unit passes through such test, we are fairly confident in that aspect of the product. 
Something in the combination of USB subsystem, cables and the camera is causing (what seems like) a power-cycle. This can be purely electrical, or caused by other drivers (any other devices on the same hub?)

dorodnic, 2018-09-04:
I can't 100% rule out the possibility of a firmware bug, so if you could share the settings you are using (resolution, fps, format of each steam and controls), @RealSense-Customer-Engineering can try to reproduce this.

goetzmi, 2018-09-04:
Hello dorodnic, at first thank you for your answer. I ordered today a new USB-Cable, which has been recommended for the D435 inside the intel forums (the cable: https://www.ntcdistributing.com/usb-3-1-type-c/a-to-c/usb-3-1-a-male-to-c-male-with-dual-screw-locking-cable/) and will test it. 

At the active USB-hub, i`m using, the D435 is the only device which is plugged in.

The settings for my device are attached:

[image]
[image]

goetzmi, 2018-09-05:
Now I`ve tested it with the depth quality tool - the same happens there after running it for an hour.

goetzmi, 2018-09-10:
Maybe I got another hint for you -> the post processing doens`t matter. The same behaviour (the librealsense::windows_backend_exception and all the other stuff) occurs without it too.

goetzmi, 2018-09-10:
And ironic white now it happened with only using the USB-cable which is delivered with the D435 (no hub, noch longer cable, without post-processing). The difference was that only the first line with the librealsense::windows_backend_exception could be found in the callstack.

goetzmi, 2018-09-14:
I think you can close this issue! But Intel should deliver better cables in future.
Im using this one: https://www.ntcdistributing.com/usb-3-1-type-c/a-to-c/usb-3-1-a-male-to-c-male-with-dual-screw-locking-cable/

RealSense-Customer-Engineering, 2018-09-14:
[Realsense Customer Engineering Team Comment]
@goetzmi,

Thank you for the update!  Happy to hear it was just the cable and nothing else.
Please let us know if you have any other questions.

closed 2018-09-14 by RealSense-Customer-Engineering