realsenseai/librealsense #2348: How to get real distance from 16-bit value in depth image

spiritas1004, 2018-09-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / **SR300** / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/**10**) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | **PC**/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I have depth images (png format), and I want to convert each pixel value (16bit) to real distance data (mm).
How can I convert it? Is there any specific formula? Thanks>

dorodnic, 2018-09-03:
Each depth sensor has a depth scale value that translates pixels to meters. 
Using it in C++ it would look something like:
```cpp
using namespace rs2;
pipeline p;
pipeline_profile profile = p.start();
device dev = profile.get_device();
depth_sensor ds = dev.query_sensors().front().as<depth_sensor>();
float scale = ds.get_depth_scale();
frameset fs = p.wait_for_frames();
depth_frame depth = fs.get_depth_frame();
uint16_t* data = (uint16_t*)depth.get_frame_data();
uint16_t pixel = data[0];
float meters = pixel * scale;
```
The syntax may differ, but the APIs are very similar in other languages.

RealSense-Customer-Engineering, 2018-09-04:
[Realsense Customer Engineering Team Comment]
Hi @spiritas1004,

The PNG image doesn't have the depth data info. You need to have the raw image (For example: use rs-convert tool to convert ros bag to raw) which data layout is according to the UVC spec for the depth stream.

WeiChihChern, 2018-09-04:
@dorodnic Will depth_scale change according to different scenes or distances between the camera and objects or it's a fixed number?

dorodnic, 2018-09-04:
It's a fixed number for sensor. With D400 advanced mode APIs you can adjust it a bit, to get better precision up close or further away, but that's a rather advanced use-case.
@RealSense-Customer-Engineering is correct, in that you need the original 16-bit depth data. Output of the colorizer is after histogram equalization, and while it does make the depth image look better, it essentially distorts all the distances.

RealSense-Customer-Engineering, 2018-09-04:
[Realsense Customer Engineering Team Comment]
This is a similar question like #2231 and #2200.
You can refer to the BKM white paper to know more about the advanced settings.
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

WeiChihChern, 2018-09-04:
@dorodnic @RealSense-Customer-Engineering Thanks for the anwer, much appreciated!

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @WeiChihChern,

Still need any support for this topic?

closed 2018-09-07 by RealSense-Customer-Engineering

jinfagang, 2018-12-17:
@RealSense-Customer-Engineering Can you help solve this issue? [https://github.com/IntelRealSense/librealsense/issues/2910](#2910 )