realsenseai/librealsense #2352: Building examples (capture) realsense D435

c3ramos, 2018-09-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | RealSense viewer 2.16.0 | 
| Operating System & Version |  Ubuntu 16.04 lTS | 
| Kernel Version (Linux Only)    | 4.15.0-33-generic                                     | 
| Platform                                 |PC |
| SDK Version                            | 2.0                        | 
| Language                            | C++                   | 
| Segment			| Building from source                   | 

I am experiencing problems when trying to build from source the examples provided by intel ( librealsense/examples at master · IntelRealSense/librealsense · GitHub  ) . 
First, I installed the realsense sdk following the instructions provided here librealsense/distribution_linux.md at master · IntelRealSense/librealsense · GitHub   and here librealsense/installation.md at master · IntelRealSense/librealsense · GitHub  , and have been able to run realsense-viewer without any issues.

Then, I tried building the capture example. To do this I created a build directory inside examples/capture and run  :

     cmake ..  -DBUILD_GRAPHICAL_EXAMPLES=true , from examples/capture/build

with the following output:

-- The C compiler identification is GNU 5.4.0

-- The CXX compiler identification is GNU 5.4.0

-- Check for working C compiler: /usr/bin/cc

-- Check for working C compiler: /usr/bin/cc -- works

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Detecting C compile features

-- Detecting C compile features - done

-- Check for working CXX compiler: /usr/bin/c++

-- Check for working CXX compiler: /usr/bin/c++ -- works

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- Performing Test COMPILER_SUPPORTS_CXX11

-- Performing Test COMPILER_SUPPORTS_CXX11 - Success

-- Performing Test COMPILER_SUPPORTS_CXX0X

-- Performing Test COMPILER_SUPPORTS_CXX0X - Success

-- Configuring done

-- Generating done

-- Build files have been written to: /home/awdw/realsense_install/librealsense/examples/capture/build

Then I run :

make

with the following output:

```
Scanning dependencies of target rs-capture
[... 349 lines cut ...]
rs-capture.cpp:(.text._ZN3rs216processing_block5startINS_11frame_queueEEEvT_[_ZN3rs216processing_block5startINS_11frame_queueEEEvT_]+0x81): undefined reference to `rs2_start_processing'

collect2: error: ld returned 1 exit status

CMakeFiles/rs-capture.dir/build.make:94: recipe for target 'rs-capture' failed

make[2]: *** [rs-capture] Error 1

CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rs-capture.dir/all' failed

make[1]: *** [CMakeFiles/rs-capture.dir/all] Error 2

Makefile:127: recipe for target 'all' failed

make: *** [all] Error 2

```
After that I checked that I had correctly installing OpenGL, finding  :
OpenGL version string: 3.0 Mesa 18.0.5

Finally, I would like to know what is missing in the procedure described above in order to correctly build from source.My knowledge of cmake is limited and I think the error must be related to that but I have not been able to find it.

I would appreciate references to any documentation where it is explained how to properly build from source.

THanks.

dorodnic, 2018-09-04:
Hi @c3ramos 
I assume you are already following [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk)

> To do this I created a build directory inside examples/capture and run :

This is the problem. For everything to work, run cmake on global CMakeList.txt in repository root. 
So, if you ran git clone under `~`, then:
```
cd ~/librealsense
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true
```

c3ramos, 2018-09-06:
Hi @dorodnic 
Thanks for your quick reply.
I followed your steps and was able to cmake and make from /librealsense/build but that command only created the binaries for rs-save-to-disk, rs-depth, rs-color and rs-distance and I was particulary interested in the **capture** examples.
Thus, I tried to run cmake ..  -DBUILD_GRAPHICAL_EXAMPLES=true  from the same directory but the Make failed.
Snippet of the output

````
[ 48%] Linking CXX executable rs-sensor-control
[ 48%] Built target rs-sensor-control
Scanning dependencies of target rs-measure
[ 51%] Building CXX object measure/CMakeFiles/rs-measure.dir/rs-measure.cpp.o
In file included from /home/awdw/realsense_install/librealsense/examples/measure/rs-measure.cpp:6:0:
/home/awdw/realsense_install/librealsense/examples/measure/../example.hpp: In member function ‘rect texture::calc_grid(float2, int)’:
/home/awdw/realsense_install/librealsense/examples/measure/../example.hpp:189:40: warning: narrowing conversion of ‘w’ from ‘double’ to ‘float’ inside { } [-Wnarrowing]
         return rect{ w, h, new_w, new_h}; //column count, line count, cell width cell height
                                        ^
/home/awdw/realsense_install/librealsense/examples/measure/../example.hpp:189:40: warning: narrowing conversion of ‘h’ from ‘double’ to ‘float’ inside { } [-Wnarrowing]
/home/awdw/realsense_install/librealsense/examples/measure/../example.hpp:189:40: warning: narrowing conversion of ‘new_w’ from ‘double’ to ‘float’ inside { } [-Wnarrowing]
/home/awdw/realsense_install/librealsense/examples/measure/../example.hpp:189:40: warning: narrowing conversion of ‘new_h’ from ‘double’ to ‘float’ inside { } [-Wnarrowing]
[ 54%] Linking CXX executable rs-measure
/usr/bin/ld: CMakeFiles/rs-measure.dir/rs-measure.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
measure/CMakeFiles/rs-measure.dir/build.make:97: recipe for target 'measure/rs-measure' failed
make[2]: *** [measure/rs-measure] Error 1
CMakeFiles/Makefile2:470: recipe for target 'measure/CMakeFiles/rs-measure.dir/all' failed
make[1]: *** [measure/CMakeFiles/rs-measure.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

RealSense-Customer-Engineering, 2018-09-18:
[Realsense Customer Engineering Team Comment]
HI @c3ramos,

Is the issue resolved based on @ev-mp's suggestion?

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
Hi @c3ramos,

I will close this issue first. You can still re-open this or create another ticket.

closed 2018-10-01 by RealSense-Customer-Engineering

zeee-yong, 2019-07-16:
Did you solve the problem???