realsenseai/librealsense #2357: Query Devices Slow If Called Twice

kevinzakka, 2018-09-05:
Give, the following imports and definitions:

```python
import pyrealsense2 as rs
import time

pipe = rs.pipeline()
ctx = rs.context()
```

running the below code more than once can take seconds to evaluate

```python
tic = time.time()
ctx_devs = list(ctx.query_devices())
devices = {}
for i in range(ctx.devices.size()):
    devices[i] = ctx_devs[i].get_info(rs.camera_info.serial_number)
toc = time.time()
print('time: {}s'.format(toc-tic))
```

Is there some resource I'm supposed to clean up that I'm not?

dorodnic, 2018-09-06:
Hi @kevinzakka 
What version of librealsense are you using?
Also, what platform?
Older versions used to return non-Intel web-cams by default, and these could take a while to get initialized. With latest version I don't see a reason for it, but I need to check if we publish the last release to be available via pip install (if that's what you are using)

kevinzakka, 2018-09-06:
@dorodnic is there a way of checking the version in python? Running `rs.__version__` throws an error.

I'm on a Macbook Pro running `10.13.3` high sierra.

RealSense-Customer-Engineering, 2018-10-23:
[Realsense Customer Engineering Team Comment]
Hi kevinzakka,

I managed to get a MacBook w/ High Sierra, and built RealSense SDK 2.13 from source code.
By running your code below, I found out that it also queried the platform camera and has some issue with it. This may consume lots of time.

If using another way to check cameras and adding time check for querying each camera, it only takes negligible time to query:
ctx = rs.context()
if len(ctx.devices) > 0:
    for dev in ctx.devices:
        print ('Found a camera: ', dev.get_info(rs.camera_info.name), ' ', dev.get_info(rs.camera_info.serial_number))

And same code running off Linux and Windows will not query platform camera.

Hope this can help.

Thanks!

RealSense-Customer-Engineering, 2018-10-26:
[Realsense Customer Engineering Team Comment]
Hi kevinzakka,

Wonder if you get any updates or do you still need help on this?

Thanks!

closed 2018-11-07 by RealSense-Customer-Engineering