realsenseai/librealsense #2361: Python+OpenCV : Close video after reading frames from bag file.

epratheeban, 2018-09-06:
I followed the code given in [read_bag_example.py
](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples). Everything is fine. But I would like to close the video after it is over. The given code is looping frames over again and again. In OpenCv python, I usually use 
```
ret, frame = cap.read()
if ret == True:
    # Do some processing 
else:
    cv2.Destroywindow()
```
So my question is does  `pyrealsense2` has any flag to identify end of the video frames ?

dorodnic, 2018-09-06:
Hi @epratheeban 
There is an optional parameter to [enable_device_from_file](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/python.cpp#L728) that control playback repeat. 
In C++ we have a `try_wait_for_frames` to then handle `wait_for_frames` timeout after last frame, but I don't see it in python right now for some reason. You can however add except to handle wait_for_frames timeout and reduce the timeout time to something more reasonable, like 50-100ms.

epratheeban, 2018-09-06:
@dorodnic  Thanks!  The wait_for_frames does the job.

closed 2018-10-18 by RealSense-Customer-Engineering

15151874182, 2019-01-02:
> @dorodnic Thanks! The wait_for_frames does the job.

@epratheeban hello,i met the same problem with yours,can i ask how did u solve this problem at last,many thanks!

soarwing52, 2019-04-05:
in enable_read_from_device I set repeat as False
and I add in the try loop one except Runtimeerror: 'print bag ended' destroy all windows()
this will close up

lghasemzadeh, 2020-10-06:
Hello guys,
I have a bag file which I recorded it via realsense viewer not ros. Now I want to read it and run an algorithm on it (bag file is the algorithm's input). My algorithms is written in python and uses OpenCV functions.
@epratheeban @dorodnic @soarwing52 @python @opencv
I would appreciate if you answer

soarwing52, 2020-10-06:
@lghasemzadeh you can follow the read bag example, which set up the pipeline and then you can read each frame with open cv as normal image and run your algorithm

lghasemzadeh, 2020-10-07:
@soarwing52 
I had tried this script before and got an error which I can't resolve it.
from script line 21, I changed it like this:
parser.add_argument("-i", "--input", default='/home/20200922_110901.bag', type=str, help="Path to the bag file")
and I get this error:
RuntimeError: Failed to resolve request. Request to enable_device_from_file("/home/20200922_110901.bag") was invalid, Reason: Failed to create ros reader: Error opening file: /home/20200922_110901.bag
How can I debug it?

MartyG-RealSense, 2020-10-07:
If you have the path to the bag correct and are getting the **pipeline.start(config)  RuntimeError: Couldn't resolve requests** message **AND** you are using RGB, then it may be worth looking at the similar-sounding situation in the case link below that involves reading RGB data from a bag with Python and OpenCV:

https://github.com/IntelRealSense/librealsense/issues/2652

lghasemzadeh, 2020-10-07:
@MartyG-RealSense 
I still getting errors.
I each bag file there are both color stream and IR streams. I never recorded and saved depth stream, I don’t want depth stream.
And my color stream is RGB8.