realsenseai/librealsense #2363: Realsense camera is not recognised.

genessisperez, 2018-09-06:
Hi, I've been having an issue with the installation of the librealsense for my cameras SR300 and D400.

Until last week, the librealsense was working until I updated the firmware from the realsense camera to latest one. Then, I started having issues with the detection of the cameras.

As I was using the ROS-wrapper, I thought I perhaps cloned a wrong version of ros-realsense and decided to update it. After that, I started having issues with ros-wrapper and complaining about incompatibility. So, I thought perhaps I needed to update also the librealsense and it was a huge mistake.

So, I tried to clean up the librealsense and the ros-wrapper, and do everything from zero. I followed first the instruction from librealsense (https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md), but once I finish following all the steps and run `realsense-viewer` or `rs-enumerate-devices`, it tells me:
` API version mismatch: librealsense.so was compiled with API version 2.14.0 but the application was compiled with 2.16.0! Make sure correct version of the library is installed (make install)`

I failed to find any error-related in the open issues at the moment, which made me think perhaps I need to build from source. So,  after removing all the SDK-packages I followed the given instructions (https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md). It builds successfully and it does not complain about any error or problem. But again, once running `realsense-viewer` or `rs-enumerate-devices`, it complains that there are no devices connected:
`No device detected. Is it plugged in?`

If I run `$ lsusb | grep 8086 `, when the camera is connected, it shows me 
`Bus 002 Device 009: ID 8086:0aa5 Intel Corp.`

The camera are plugged using a USB 3.0, and the cameras work in different computers. Just not in mine.

The result from `$ uname -r ` is:
`4.15.0-33-generic`

and I'm running on Ubuntu 16.04.1.

I don't know what to do anymore, as I've tried almost every single possible option that I've read in your Q&A and in another issues. Could someone help me?

Thanks and cheers!

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @genessisperez,

1. Per your USB VID/PID = 8086:0aa5, it's a SR300 device. Please run "usb-devices" to show me more info about the USB connection status.
2. What's your librs version and firmware version for SR300 and D400?
3. Find the proper ros wrapper version in https://github.com/intel-ros/realsense/releases
4. And, you can refer to below similar issue which is due to the installation and build process to cause the API mismatch.
https://github.com/intel-ros/realsense/issues/442

genessisperez, 2018-09-07:
Here's the output for the SR300:

`T:  Bus=02 Lev=02 Prnt=06 Port=00 Cnt=01 Dev#=  7 Spd=5000 MxCh= 0`
`D:  Ver= 3.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1`
`P:  Vendor=8086 ProdID=0aa5 Rev=31.50`
`S:  Manufacturer=Intel(R) RealSense(TM) Camera SR300`
`S:  Product=Intel(R) RealSense(TM) Camera SR300`
`S:  SerialNumber=614204000271`
`C:  #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=0mA`
`I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=(none)`
`I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo`
`I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=(none)`
`I:  If#= 3 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo`
`I:  If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)`
[usb_realsensesr300.txt](https://github.com/IntelRealSense/librealsense/files/2360636/usb_realsensesr300.txt)
Firmware version: 3.21.0.0

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @genessisperez,

Per your usb device info, the uvcvideo may not loaded normally. Try to run below commands after re-connecting to SR300 or D435.
several checks:
1. "modinfo uvcvideo | grep ver"
2. dmesg
3. follow below instructions, try again to install the pre-built DKMS package, run "sudo apt-get install librealsense2-dkms", reboot your PC, then see how it works.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

genessisperez, 2018-09-10:
Hi @RealSense-Customer-Engineering 

Seems like problem solved! Thanks a lot!

closed 2018-09-10 by genessisperez

likestudy, 2019-11-12:
@RealSense-Customer-Engineering Hello, I meet the same problem, And i have tried on 4 computers by following https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.
After the step "sudo apt-get install librealsense2-utils",  the information by "modinfo uvcvideo | grep "version:" is:  
version:        1.1.2.realsense-1.3.5
srcversion:     35CF6FF13E7340591034B90
And the output for "usb-devices" is :
[output_realsense.txt](https://github.com/IntelRealSense/librealsense/files/3835857/output_realsense.txt)

T:  Bus=04 Lev=02 Prnt=03 Port=00 Cnt=01 Dev#=  4 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=8086 ProdID=0aa5 Rev=31.10
S:  Manufacturer=Intel(R) RealSense(TM) Camera SR300
S:  Product=Intel(R) RealSense(TM) Camera SR300
C:  #Ifs= 5 Cfg#= 1 Atr=c0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 2 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
I:  If#= 3 Alt= 0 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
[output_realsense.txt](https://github.com/IntelRealSense/librealsense/files/3835895/output_realsense.txt)

And i tried : 
1\"modinfo uvcvideo | grep ver" : 
version:        1.1.2.realsense-1.3.5
description:    USB Video Class driver
srcversion:     35CF6FF13E7340591034B90
vermagic:       4.10.0-28-generic SMP mod_unload 
2\dmesg
[   45.950452] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950479] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950489] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950498] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950516] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950525] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.950533] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   45.978527] ACPI Warning: \_SB.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20160930/nsarguments-95)
[   46.111905] nvidia-modeset: Allocated GPU:0 (GPU-5dae5250-fe36-47a7-863f-b9ba3a2f2010) @ PCI:0000:01:00.0
[   46.112114] nvidia-modeset: WARNING: GPU:0: Correcting number of heads for current head configuration (0x03)
3\sudo apt-get install librealsense2-dkms 
It shows the librealsense2-dkms is the newest version 1.3.5-0ubuntu1
4\reboot
After reboot and "realsense-viewer", still no picture here.

As i tried this way in many computers, i think there maybe some steps that i missed. But only few steps to open the realsense-viewer.
The system is Ubuntu16.04, with 4.10.0-28-generic.
Any advices will help me a lot. Thanks.

ev-mp, 2019-11-12:
@likestudy hi,  please provide the details and logs for assessment -
What do you mean by no picture ? Is the camera being recognized by viewer, but you cannot start streaming ?
 Which sensors did you try to activate and what were the errors you've received ?
Does the device passes enumeration correction ? What does the kernel log show when you plug-in the camera?
Can you collect and provide the realsense application console outputs and logs for
 - `rs-enumerate-devices -c`
 - `rs-depth`
 - `rs-capture`

P.S: Please open a new ticket for support

likestudy, 2019-11-12:
@ev-mp Sorry for that i did not describe my problem clearly. I open a new issue that  [Realsense camera can not be recognised](https://github.com/IntelRealSense/librealsense/issues/5230).
Thanks for your help.

a26413768, 2020-09-19:
> [Realsense Customer Engineering Team Comment]
> Hi @genessisperez,
>
> Per your usb device info, the uvcvideo may not loaded normally. Try to run below commands after re-connecting to SR300 or D435. [...]

Thanks a lot! you save my day