realsenseai/librealsense #2365: R200 failing to launch with ROS

DMG22, 2018-09-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  R200 | 
| Firmware Version                   | 1.0.72.10 | 
| Operating System & Version |   Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  4.15.0| 
| Platform                                 | PC |
| SDK Version                            |  legacy                          | 
| Language                            |                           | 
| Segment			|                        | 

### Issue Description

Hi,

I am trying to use an R200 camera with ROS kinetic on Ubuntu 16.

Whenever I try to launch the ROS launch file I have a problem where the ROS nodelet manager dies.

Does anyone have experience with ROS that knows what is going on here?

[FATAL] [1536299039.094097024]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/nodelet_manager-1] process has died [pid 9843, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/dylan/.ros/log/85513740-b256-11e8-b0e5-989096a21409/camera-nodelet_manager-1.log].
log file: /home/dylan/.ros/log/85513740-b256-11e8-b0e5-989096a21409/camera-nodelet_manager-1*.log
[camera/driver-2] process has died [pid 9844, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/dylan/.ros/log/85513740-b256-11e8-b0e5-989096a21409/camera-driver-2.log].
log file: /home/dylan/.ros/log/85513740-b256-11e8-b0e5-989096a21409/camera-driver-2*.log

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
Hi @DMG22,

Please review below wiki and consider to build from SOURCE. You should be able to run the R200 launch file on ROS Kinetic + Ubuntu 16.04.
http://wiki.ros.org/realsense_camera

DMG22, 2018-09-14:
Hi,

This worked! I successfully installed and now when I run the launch file there are no errors and the camera feed is visible in Rviz.

Thank you very much!

closed 2018-09-14 by RealSense-Customer-Engineering

FPSychotic, 2019-01-24:
> Hi,
> 
> This worked! I successfully installed and now when I run the launch file there are no errors and the camera feed is visible in Rviz.
> 
> Thank you very much!

im new in ros, i dont know how to setup the .launch for the r200.
can you share yours please
in the launch, do I need a cam_name? cam0_topic is a constant or i need put /camera