realsenseai/librealsense #2366: Exception realsense2.dll after long term running.

goetzmi, 2018-09-07:
Required Info |  
-- | --
Camera Model | D435
Firmware Version | 5.10.3.0
Operating System & Version | Windows 10 (x64)
Kernel Version (Linux Only) |  
Platform | PC
SDK Version | 2.15.0.163
Language | C++/CLI
Segment | Robot

### Issue Description

After the problems I`ve posted here in Connection loss after long term running. (<=2h) #2345
I started to do long term tests with only the (way to short) cable which is delivered with the
D435. After some time (last run after 2,7h)  I get the following exception:

[image]

Unfortunately I don`t manage to chatch it. Here`s the callstack and disassembly:

[image]

[image]

After this exception I try to replug the camera, and when I do that the depth image tool
shows incomprehensible way that the device is connected to a USB 2.1 port and in due
to that it`s no longer possible to run the full resolution. Only after some replugs the correct
port version (3.2) is shown:

[image]

The settings of the device are still the same as I used here: 
https://github.com/IntelRealSense/librealsense/issues/2345

I hope that this device will run stable one day - because we planned to order it in a bigger
amout after our testapplication works nice with this device.

MartyG-RealSense, 2018-09-12:
You may be able to reduce the likelihood of the camera being detected in USB2 mode if you plug the camera in quickly and firmly instead of slowly.  A better way to avoid resetting the camera with unplug-replug though is to use a hardware_reset() instruction at the start of a script.

https://communities.intel.com/message/557115#557115

The discussion linked to below about long-term running may also be of use to you.

https://communities.intel.com/message/560215#560215

MartyG-RealSense, 2018-09-13:
I'm very glad that the connection-loss aspect of your problems is resolved by your cable choice.

Long-term running was mentioned in an online RealSense webinar that Intel held yesterday.  It was said that the camera should be able to run indefinitely as long as it remains within temperature tolerances, and problems that disrupt running are usually related to an issue happening on the PC, such as a USB glitch.

Looking at the ASIC and projector temperature readings in the image provided in the other post, I noticed that the ASIC chip seems to be running a bit hot at 43.   With a test of my own camera, running at high resolution, it was running the ASIC at a temperature of 30 at cold startup.   

In the 400 Series' data sheet document, the maximum recommended 'operating temperature' is 35 degrees.  So if the values in the Viewer are in degrees, then I could foresee the possibility for glitches setting in if this vital chip is running at 43 degrees after some hours of operation.

In the webinar session during discussion about long-running, it was recommended that a large heat-sink component is purchased (the heat sink being the part that draws heat away from the circuit board).  In choosing a heat sink, it was suggested to base the decision on the maximum power output of the USB specification, in order to be able to cope with worst-case scenarios.  

Installing a custom heatsink may be easier for the caseless Module Kit versions of 400 Series cameras than it is for the D435, which is of course a cased unit and not supposed to be opened up.  So with the cased units, long-run stability relies on keeping the camera within temperature tolerances and making the PC part of your setup as reliable as possible.

goetzmi, 2018-09-13:
Hello MartyG,

I`m trying to observe the temperate with different cooling options I got. But unfortunately
I get the rs2::wrong_api_call_sequence_error in codeline "depth_sensor_temperature = rs2_get_option((const rs2_options*)sen..". What I am doing wrong? I do this after initializing the device and every time before grabbing a frame:

```bool is_depth_sensor_found;
float depth_sensor_temperature;
rs2_sensor_list *sensor_list = rs2_query_sensors(dev, &e);
int num_of_sensors = rs2_get_sensors_count(sensor_list, &e);

for (int i = 0; i < num_of_sensors; i++)
{
	rs2_sensor* sensor = rs2_create_sensor(sensor_list, i, &e);
	is_depth_sensor_found = rs2_is_sensor_extendable_to(sensor, RS2_EXTENSION_DEPTH_SENSOR, &e);

	if (is_depth_sensor_found)
	{
		depth_sensor_temperature = rs2_get_option((const rs2_options*)sensor, RS2_OPTION_ASIC_TEMPERATURE, &e);
		cout << "Temperatur ASIC [" << i << "]: " << depth_sensor_temperature << endl;
	}

	rs2_delete_sensor(sensor);
}

rs2_delete_sensor_list(sensor_list);

goetzmi, 2018-09-14:
Hello MartyG,

since yesterday I cooled the camera down with a fan and held the temperature constant at 31°C.
But after 19hours 37minutes the exception in realsense2.dll occured again (from now 10 minutes ago).
The PC I`m using is a Beckhoff industrial device with Win10 (x64) and should be very stable - we`re
using them since more than 10 years (always the up to date version).

MartyG-RealSense, 2018-09-14:
I guess you mean you have been using Beckhoff industrial PCs for more than ten years, and not that you've been using the same PC for 10 years.  :)

I read back through this case from the start.  I see your error with the RealSense2.dll is occurring in an application called NativeApplicationTest.exe.  Is this a custom application that you have written yourself to test the camera?

The next diagnostic step I would recommend would be to monitor your PC's memory usage whilst the program is running.  This will tell you whether or not a 'memory leak' in your program is causing instability and eventual program crash.  A memory leak is a bug in a program where the memory is not being released correctly when an operation in a process closes.  Over time, more and more of the PC's memory is consumed until the program can no longer run.

goetzmi, 2018-09-14:
Youre right, of course we are not using the same PC for more than ten years. But this one we are using now for about 1,5 years in other vision applications and with other cameras (Ethernet, USB, Firewire) without any problems. 

The application is written by myself. For the long-run tests it only acquires images from the D435 and
visualizes it in a window and updates a clock for time measuring as well as the memory used by the application. The PC has 32GB DDR4 RAM and the usage (6,9GB for the whole system and my application) is constantly stable during the run - there is no memory leak, otherwise the amount of used memory would rise (as you still wrote) or an exception like AccessViolationException or OutOfMemoryException would be thrown.

The energy saving options of the system are all off, so that it isnt able to turn something like a USB port off. The D435 is 5m from the PC away (therefore the 5m USB-cable).

MartyG-RealSense, 2018-09-14:
The causes that have been ruled as unlikely to be the source of the error include:

- Temperature
- PC stability (presumably including USB ports if it is an industrial PC)
- Memory leaks

It is also probably not a corrupt Windows registry generating the exception, as the exception error is usually displayed when you try to launch the program if the registry is at fault, rather than after the program has been running for hours.

Your application does not sound like it is stressing the PC and camera too much, so using an industrial-grade USB cable may not make a difference in this case (industrial-grade USB equipment is used in critical applications where frame drops can not be allowed, such as medical scanners).

From my knowledge of past cases involving long-running, typically the behavior described (the program stopping after half a day instead of days) happens when there are multiple cameras being used instead of a single camera.   Since you are only using a single camera, that raises the question of whether something else (e.g other devices) are using the USB ports too and putting a similar strain on the system to that which multiple cameras place.

agrunnet, 2018-09-14:
This may be somewhat related. I don’t know but I thought I would link to it here. The fix in that case for the Long cable was a new USB3 cable. 

https://github.com/IntelRealSense/librealsense/issues/2345#issuecomment-421264104

MartyG-RealSense, 2018-09-14:
I remember dealing with a case where I recommended a locking cable that screwed into the little holes either side of the camera's USB port on its side.  In that case, a user in Germany had a mobile robot I believe, and was losing connection because the USB cable at the camera end was coming loose.

goetzmi, 2018-09-17:
Hi MartyG,

thank you for the answers! I still saw and read your linked postings - the same behaviour I got with the cable which is delivered with the D435. With the new one its gone.

But I dont get USB-disconnects anymore - only the realsense2.dll error we wrote about. The camera doesnt loose connection (since I got the new cable) when the exception rises. The only devices which
are plugged to the PCs USB-ports are keyboard and mouse - Ill will disconnect all other devices for
the next test.

@agrunnet: It was me that reported the error you linked ;D.

MartyG-RealSense, 2018-09-17:
Thanks goetzmi for the feedback.  A keyboard and mouse may not place much demand on the USB ports in the same way that a high-power device such as an external hard drive would.  It may be worth doing the test anyway, since my own PC is plagued with USB mouse disconnects-reconnects recently (I suspect my elderly USB hub as the cause, as it happens with all mice).

It occurs to me that another cause of a camera disconnecting over time is if it goes to sleep due to inactivity because of power-saving settings in Windows. 

 In the older generations of RealSense camera, if no input was being made to the keyboard and mouse then the camera could go to sleep even if it was in active use if power-saving settings permitted sleep.  This was because there was no input that Windows cold recognize as 'activity', unlike a keyboard key press or a move of the mouse.  So you could be continuously waving your arms in front of the camera and it could still go to sleep if power-saving was enabled and there was no touch of the mouse or keyboard.

MartyG-RealSense, 2018-09-17:
I'm not a Visual Studio user so my apologies if this suggestion is not practical: would it be possible to avoid the exception crash by turning off exception throwing in Visual Studio? It's a 'dirty' way of developing (error messages are useful, if annoying) but if the program is otherwise working fine, disabling exceptions in your project mat be okay.

https://social.msdn.microsoft.com/Forums/vstudio/en-US/13b1b4cb-4efa-47e3-901e-375dfa63988f/how-to-turn-off-quotbreak-when-exception-is-thrownquot-option-for-custom-exception?forum=vsdebug

goetzmi, 2018-09-17:
Thank your for your suggestion, but this is the last option Ill do - in due to the reason its "dirty".
The x64 build now succeed and Ill attach to the process with the VS solution of it. Maybe Ill get the trigger of this behaviour this way.

goetzmi, 2018-09-18:
Okay, now I got the SDK builded and all symbols of it loaded into my application. Lets see what causes the error - usually in 20h we will know more.

MartyG-RealSense, 2018-09-18:
At times like this, you wish for a time machine so you can skip ahead.  *smiles*   Looking forward to hearing your results!

goetzmi, 2018-09-18:
Oh yes! You can say that again!

goetzmi, 2018-09-19:
Since today I had two exceptions. The first Ill mention in the lines below does stop my application and I had to restart it. The exception raises inside the synthetic_source class, which Im not using as far as I know:

**- Exception:**
[image]

**- Callstack:**
[image]

**- Codelines where it happens:**
[image]

The second exception doesnt stop the application - it goes on executing after ignoring it
or tell the debugging to continue. It happens in a private function of the align class and
uses private member - not a single one thats given to from my code.

**- Callstack:**
[image]

**- Codelines where it happens:**
[image]

MartyG-RealSense, 2018-09-19:
Hopefully the Intel team can offer some advice on this one, as it requires a depth of familiarity with the SDK code that I don't have.

goetzmi, 2018-09-19:
3.) Finally I got 5 minutes ago the exception with the same callstack output which I initially mentioned with this issue:

**- Exception:**
[image]

**- Callstack:**
[image]

**- Codelines where it happens:**
[image]

Hopefully the Intel team will look at this soon!

goetzmi, 2018-09-19:
If this is an option for the intel team - I can offer that they connect to the system Im using.
I let it now run until the exception rises again and will stop it then there, so that (just if they want)
can have a closer look to it.

RealSense-Customer-Engineering, 2018-09-21:
[Realsense Customer Engineering Team Comment]
Hi @goetzmi,

Per #2345, you don't see the USB disconnect issue with the proper USB cable. Using realsense-viewer (the latest, ex: librealsense 2.16.0) also has this exception issue?
Per your new long-run test, does all image processing happens on the graphics?
Can you enable the realsense log file (export LRS_LOG_LEVEL=DEBUG) and once the exception happens, dump the dmesg?
You can still continue monitor the temperature based on below code:

 auto dbg = dev.as<rs2::debug_protocol>();
 std::vector<uint8_t> cmd = { 0x14, 0, 0xab, 0xcd, 0x2a, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };
 auto res = dbg1.send_and_receive_raw_data(cmd);
 int temp = res1[4];
 std::cout << "Device Temperature: " << temp << std::endl;

goetzmi, 2018-09-21:
'Using realsense-viewer (the latest, ex: librealsense 2.16.0) also has this exception issue?'
The viewer just stops working, but doens`t show the exception in a dialog when this happens.

'Can you enable the realsense log file (export LRS_LOG_LEVEL=DEBUG) and once the exception happens, dump the dmesg?'
-> Of course, where and how can I enable the log file?

Thank you for the code!

MartyG-RealSense, 2018-09-26:
I'm very sorry you have been kept waiting for a further response.  A new version of the SDK (2.16.1) was released yesterday.  Would it be possible to try it and see whether your problems persist.  Thanks!

https://communities.intel.com/thread/129313

goetzmi, 2018-09-26:
Of course.

goetzmi, 2018-09-26:
The first thing I noticed with the new Version, is that the following message now pops up every second and ist the same which I initially posted here:

[image]

If I ignore it and force the debugger not to stop at this exception the application runs well. But thats not what I mean by a clean solution.

MartyG-RealSense, 2018-09-26:
If you can run for long periods (3 days, for example) when the exception is turned off and experience no ill effects, the warning message is probably not worth worrying about.  Sometimes you can't fix everything.

goetzmi, 2018-09-26:
Lets see what happens during the next days..

goetzmi, 2018-10-15:
Hello MartyG. The device now runs stable - it`s not very nice with the exception which the realsense2.dll throws in background but it`s doing its job if you ignore it in VisualStudio2017.

Little summary of the solution steps for everyone who runs into this behaviour too:

1.) Get this cable to avoid connection losses in due to LTE-frequences and the cable lenght:
https://www.ntcdistributing.com/usb-3-1-type-c/a-to-c/usb-3-1-a-male-to-c-male-with-dual-screw-locking-cable/

2.) Build some cooling around the camera for long duration runs to hold it below 35°C. I used
a simple 12V powered one.

3.) The current SDK-Version seems to handle some error which still occurs, but is now catched.

In my oppinion you can close this thread - thank you very much for your help and patience!

Best regards,

goetzmi

MartyG-RealSense, 2018-10-15:
Awesome news!  Thanks so much for your own patience and repeated testing.  I'm really glad that you achieved satisfactory results.  Thanks a lot too for sharing your solution method    :)

goetzmi, 2018-10-16:
Here I got some realsense logfiles for you - maybe they`re helpful:

[2018-10-09-08_24_48.log](https://github.com/IntelRealSense/librealsense/files/2482747/2018-10-09-08_24_48.log)
[2018-10-12-08_23_57.log](https://github.com/IntelRealSense/librealsense/files/2482748/2018-10-12-08_23_57.log)
[2018-10-12-13_17_50.log](https://github.com/IntelRealSense/librealsense/files/2482749/2018-10-12-13_17_50.log)
[2018-10-15-08_28_04.log](https://github.com/IntelRealSense/librealsense/files/2482750/2018-10-15-08_28_04.log)
[2018-10-11-13_31_18.log](https://github.com/IntelRealSense/librealsense/files/2482751/2018-10-11-13_31_18.log)
[2018-10-11-13_31_45.log](https://github.com/IntelRealSense/librealsense/files/2482752/2018-10-11-13_31_45.log)

MartyG-RealSense, 2018-10-16:
Thanks again!!   :)