realsenseai/librealsense #2367: Unity crash, AlignmentSample scene

PaulBre, 2018-09-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.10.3 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.16.0 | 
| Language                            |  Unity  | 
| Segment			| VR | 

### Issue Description
When I run the Unity scene AlignmentSample Unity (2018.2.5f1) crashes with error log:

realsense2.dll caused an Access Violation (0xc0000005) in module realsense2.dll at 0033:02802272.

Stack Trace of Crashed Thread 12516:
0x00007FF916062272 (realsense2) rs2_wait_for_frame
0x00007FF916061293 (realsense2) rs2_wait_for_frame
0x00007FF91606595C (realsense2) rs2_wait_for_frame
0x00007FF9160712CA (realsense2) rs2_wait_for_frame
0x00007FF916072B23 (realsense2) rs2_wait_for_frame
0x00007FF916072AA0 (realsense2) rs2_wait_for_frame
0x00007FF91604821D (realsense2) rs2_process_frame
ERROR: SymGetSymFromAddr64, GetLastError: 'The specified module could not be found.' (Address: 00000000405A3EAF)
ERROR: SymGetModuleInfo64, GetLastError: 'A dynamic link library (DLL) initialization routine failed.' (Address: 00000000405A3EAF)

When I open the scene PointCloudDepthAndColor, activate RsAlign in the RsProcessingPipe object and run it the Unity Editor crashes as well (with the same error log).

For my project I realy need a color steam that is aligned with the depth data. Please help.

frenchtouchXR, 2018-09-14:
Hello, the same happens when I playback a recorded file (on Unity) since 2.15 version. I think the new pipeline needs some improvements.

AleBasso80, 2018-09-25:
Hello,
I get a similar error in C++ under Linux that I didn't get with the previous versions.

RealSense-Customer-Engineering, 2018-10-01:
[Realsense Customer Engineering Team Comment]
Hi AleBasso80,

As for C++, you can review the latest rs-align sample code to see how it works.
As for Unity, "AlignmentSample.unity" scene with rsAlign script can reproduce this issue.

AleBasso80, 2018-10-01:
Hi,
I found out that my code works fine until the 2.15 release. (Huge) Problems arise with the 2.16.0 version.

RealSense-Customer-Engineering, 2018-11-20:
[Realsense Customer Engineering Team Comment]
Please try the latest librealsense to see how it works. It should be fixed.

closed 2018-11-28 by RealSense-Customer-Engineering