realsenseai/librealsense #2372: How to use align in python for other RGB source

Steven0706, 2018-09-10:
I am using window 10 with D435 and most recent FW (09092018). I do not know C++ much so I use the python wrappers. My question is simple, how to use the align function provided in the python example on other RGB camera source? Say I have a webcam which is in very close distance of D435. I use opencv to read the frame from the webcam. something like:
cap = cv2.VideoCapture(0)
_, frame = cap.read()

I also see the python example of the realsense align-depth2color:
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
align_to = rs.stream.color
align = rs.align(align_to)

My question is, how do I feed my opencv webcam source to this re.align function that I can get my RGB with depth info aligned? Thank you so much! :)

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
Hi @Steven0706,

Are you able to calibrate your RGB sensor with our Left sensor in order to have accurate alignment? Or, you can roughly have a rotation/translation transform based on the mechanical design?

RealSense-Customer-Engineering, 2018-11-28:
[Realsense Customer Engineering Team Comment]
Hi @Steven0706,

still need any support

closed 2018-11-28 by RealSense-Customer-Engineering