realsenseai/librealsense #2373: Bad display of disparity map on Android application

fotsing365, 2018-09-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu 16.04| 
| Kernel Version (Linux Only)    |  (4.10.0)                                         | 
| Platform                                 | Android |
| SDK Version                            |               | 
| Language                            |  {C++ }                          | 
| Segment			|  Rooted Lenovo TB-8504F                          | 

### Issue Description
< when I launch the application the disparity map displayed is really bad. Please does anybody know how I could improve the result?
[image]
>

RealSense-Customer-Engineering, 2018-10-17:
[Realsense Customer Engineering Team Comment]
Please take a picture of your setup, including the RealSense camera and the scene.
What is your firmware version?

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi fotsing365,

Is there any progress? Is this still an issue?

closed 2018-11-05 by RealSense-Customer-Engineering