realsenseai/librealsense #2379: rs2::error

geoffzhang, 2018-09-11:
Hi，when I ran the sample, I had an error which is "rs::error". How can I solve it, please. @chachi @wspr @martinus @rojkov

dorodnic, 2018-09-12:
Hi @geoffzhang 
What sample did you run?
Usually the error content is printed out to console. What was the output? 
Also, to find the root cause of the issue, we will likely need OS (type and version) and camera type.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
@geoffzhang Could you please provide more details about the issue?

1. What sample did you run to get this issue?
2. Operation system information
3. librealsense and camera FW version
4. Camera type

geoffzhang, 2018-10-19:
Thanks, I have solved this issue, I update my windows system, then it worked
.

geoffzhang, 2018-10-19:
Thanks, I have solved this issue, I update my windows system, then it worked
. @dorodnic @RealSense-Customer-Engineering

RealSense-Customer-Engineering, 2018-10-22:
[Realsense Customer Engineering Team Comment]
@geoffzhang Thanks for the update! We'll close this ticket accordingly.

closed 2018-10-22 by RealSense-Customer-Engineering